Ultimaker/Cura #7976: 3d Printing, ADHESION, better method of removal possible??

Reality4DEvolution, 2020-06-24:
I am a new 3d printer user.  but i do well on learning curves.
PROBLEM - adhesion of plastic, to the bed.  CURA has couple options to adhere stuff - BRIM, and RAFT
BRIM helps to ADHERE stuff, but is difficult to remove, to get scraper underneath, and start the removal.

RAFT uses a lot of plastic, and it also hs a problem to remove;  the objext itself can break away from the raft, but raft difficult to scrape off.

I have a workaround, and it uses LESS plastic, and is easier to remove, AND helps adhere stuff to the bed.  For simplicity, call it WINGS.  PERHAPS cura can develop a way to use THIS, as an alternative, if you wish.

the WINGS are small (about 1 to 2 mm) widths of flat object, that is TAPERED OUTWARD FROM the bed to the top, to allow a small triangle to get teh scraper underneath, to start loosening the object.  I have them at about 5 mm width, and about 10 mm length, but they could be altered (similar to BRIM).

I have them in SKETCHUP, as a favorite, and i can import them to my projects, and attach them to teh sides at the bed level.  ON CIRCLES, these can actually be moved INTO the circumference, so there is actually plastic within plastic.

See attached , i used a thingiverse wheel bearing, but added WINGS to it, before printing.  They are easy to cut off once printed.  Als included are the SKETCHUP file and the STL file, to give examples.

In Sketchup, they are separate components, so they stay as a unit, and make it easy to move them around and attach.

Anyway, just something I've found useful, saves plastic, helps removal from bed.  Feel free to consider adapting this, or include it as a suggestion, etc, in future development.

and keep up the good work.

NOTE, your system does not accept the STL or SKP file, so they are in this dropbox link for the next several weeks
https://www.dropbox.com/sh/m5zeb90d75xemt1/AABXNblwjpLOsTqHZXwxmt08a?dl=0

and here is a JPG of this
[image]

Ghostkeeper, 2020-07-01:
This is in many ways similar to the [mouse ears](https://github.com/Ultimaker/Cura/issues/1349) feature request.

It's a little problematic when you're dealing with arbitrary shapes. Like if there isn't really a broad flat-ish shape to adhere to:
[image]
These are 1mm-radius pillars.

Another problem is that you need to decide where those wings go. In my case I can't just make a wing on every part on the most extreme X coordinate because that would make them intersect with the part next to it. Deciding that sort of thing is something that humans are good at, but computers not so much. It can certainly be done, but it becomes a matter of trying out every possible location and seeing whether the wing collides with anything then. Sometimes there will not be a solution.

> NOTE, your system does not accept the STL or SKP file, so they are in this dropbox link for the next several weeks

Github accepts them if you put them in a .zip :)

GregValiant, 2026-09-08:
These are now available with one of Slashee-the-cows plugins in the Marketplace.  I'll go ahead and close this.

closed 2026-09-08 by GregValiant