Ultimaker/Cura #7977: Cura 4.6.1 Prime Tower - Skipping layers for secondary extruder.

hinotori74, 2020-06-25:
Hi all,

I use the second extruder to print the supports (PVA - Support Roof Extruder/Support Floor Extruder), but when generating the gcode, it is observed that the printing of the second extruder is not constant in the tower, so it does not have a base on which to print.

[image]

I understand this may be to save time and material, but the first extruder must at least fill that space for the second extruder to have a base to print on.

Sorry for my English, i speak spanish.
Thanks

BagelOrb, 2020-06-26:
@hinotori74 please use the issue template when filing a bug report.

BagelOrb, 2020-06-26:
This is a feature, not a bug. This is intended behavior.

We think that a prime tower actually prints well like that. Do you have a photo of a failed prime tower because of this issue?

Ghostkeeper, 2020-07-02:
Duplicate of https://github.com/Ultimaker/Cura/issues/6909.

closed 2020-07-02 by Ghostkeeper