Ultimaker/Cura #7978: With Combing Mode On, Z lift for next layer is not at the end.

markkahler, 2020-06-25:
**Application version**
Cura v4.6.1

**Platform**
Windows 10

**Printer**
Self built.

My printer uses 0.8 -> 1.4mm nozzles.
Due to the nozzle size, it is prone to oozing and thus leaving blobs on the print surface.
To avoid/reduce this from happening, I now print with Combing Mode: 'All'.
What I have discovered thou is that the Z height change for the next layer is happening at the start of the final ";MESH:NONMESH" section and not at the end.
This is causing the nozzle to lift prior to the final moves to the next seam position and thus I get blobs deposited on the print.

**Actual results**
G0 F1800 X289.58 Y223.274
G1 F2100 E-4
;MESH:NONMESH
G0 F600 X289.58 Y223.274 Z1
G0 F1800 X291.297 Y223.552
G0 X292.77 Y224.087
.
.
.
G0 X324.386 Y264.888
G0 X326.635 Y266.258
;TIME_ELAPSED:163.917808
;LAYER:1
M106 S255
;TYPE:WALL-OUTER
;MESH:covid19_headband_rc3.stl
G1 F2100 E4.41575
G1 F1620 X467.089 Y258.003 E0.02844

**Expected results**
G0 F1800 X289.58 Y223.274
G1 F2100 E-4
;MESH:NONMESH
G0 F600 X289.58 Y223.274
G0 F1800 X291.297 Y223.552
G0 X292.77 Y224.087
.
.
.
G0 X324.386 Y264.888
G0 X326.635 Y266.258 Z1
;TIME_ELAPSED:163.917808
;LAYER:1
M106 S255
;TYPE:WALL-OUTER
;MESH:covid19_headband_rc3.stl
G1 F2100 E4.41575
G1 F1620 X467.089 Y258.003 E0.02844

Ghostkeeper, 2020-07-02:
Cura does this on purpose. Since Cura 3.4-ish it's moving to the Z coordinate of the next layer before moving to the X/Y location of the starting point, in order to reduce wiping off the nozzle on the previous layer or scraping that layer. If we implement what your expected results are, it'll scrape the previous layer even harder, resulting in bigger blobs. Why would you think that results in less blobs?

Might I suggest to enable retraction on layer switch, and perhaps to enable Z hops?

markkahler, 2020-07-02:
With Combing On, the nozzle is already wiping off/scraping on the just printed layer for the whole time that layer was printed prior to the last move in question. So I don't see why lifting for the last move only, will make any difference to wiping or scraping of the nozzle on the just printed layer.

Prior to me editing the G-Code and manually moving the Z height change to the end of the layer code, my 0.8mm nozzle left several blobs on the just printed layer. These caused the nozzle to jump/wobble when it moved over them during the next later print and this was then very visible in the print.
After the G-Code edit, this no longer happened and I got a much better result.
Retraction and Z hop did improve things a little, but not as much as keeping the nozzle down.

With the nozzle remaining at the same Z height till the very end of the layer, any oozing is ironed flat by the nozzle at the same time, so the oozing will not have a major effect on the next layer to be printed.

Ghostkeeper, 2020-07-06:
> With Combing On, the nozzle is already wiping off/scraping on the just printed layer for the whole time that layer was printed prior to the last move in question.

The bug was about the travel move towards the starting location of the next layer though, wasn't it? With combing enabled, it'll move through the inside of the model without needing a retraction (and thus without Z hop). This is a way to save on printing time, because blips on the inside of the model are not visible anyway. If you don't want that, disable combing. That's unrelated to what I thought this issue was about.

I thought you were talking about the travel move to the next layer being like this:
[image]

Instead of this:
[image]

Then I'd say that the current result is better than your expected result, because your expected result would scrape over the previously printed layer more. Especially if you imagine in these drawings some other object or wall in the middle.
Indeed, if you're lucky it would move completely over 100%-density infill or skin and then keep the counterpressure on the nozzle. But if you have sparse infill or multiple parts it would give a worse result than the current behaviour. Sparse infill and multiple parts are both very common.