Ultimaker/Cura #7979: Absolute Extrusion reset per layer.

markkahler, 2020-06-25:
I have needed to edit my G-Code to add or reduce some extraction amount on a layer.
In order to do this, I have needed to use the option 'Relative Extrusion' 'On'.
The issue I then have is with Relative Extrusion, the final print results are not a good/smooth as the results when using 'Absolute Extrusion'

A possible solution would be to have the option to reset the Absolute Extrusion after every layer, thus I could 'swap' the required layer G-Code with Relative Extrusion G-Code.

Or better still, have the option with 'Relative Extrusion' to select what layers only need to be changed to Relative and the rest remain Absolute.

nallath, 2020-06-29:
> I have needed to edit my G-Code to add or reduce some extraction amount on a layer.

Could you explain a bit more what you mean by this? A number of post-processing scripts might already do what you want. They take the inverse approach for this; They keep everything absolute and switch to relative, do something and then switch back to absolute.

markkahler, 2020-06-29:
Hi Jaime,

I was involved with a group called 3D Croud, printing the Prusa RC3 Face 
Shield for key workers at the start of the Covid-19 pandemic.

The printer I was using was one I designed and built incorporating a E3D 
Super Volcano HotEnd.
I was printing the Shield frames with a 0.8mm nozzle, 0.5mm layer height 
and a line width of 1.2mm - for faster printing and good quality.

The problem I was having was that after printing certain layers - like the 
top layers of the studs to hold the shield film and the top layers for the 
elastic holder.
After retraction at these points and then travel to the next start of 
print, depending on the distance of travelled, the nozzle would ooze a bit 
(due to the size) and then not start to print at the exact start of next 
extraction point.
I had already added a 'Retraction Extra Prime Amount' to the Cura slicer, 
that was just enough to not over extrude on short hops, but this was not 
enough for long hops and so a small 'hole' would be left in the print.
As a result I used Relative Extrusion mode and would then edit the G-Code, 
find the long hop moves and manually add an extra Prime amount to 'fix' 
the problem.
I also added some extra extortion to a few areas on the 1st layer, to make 
a more solid layer/close holes.

This worked well for closing the 'holes' but as a result of having to 
print the whole model in Relative Extrusion mode meant the outer skin was 
not as smooth as when printed with Absolute Extrusion.

So as a thought to get 'the best of both worlds' I suggest resetting the 
extrusion counter after every layer in Absolute Extrusion mode, thus 
making it is possible to mix Absolute and Relative Extrusion within the 
same print job.
Or better still, have the option to pick what layers you would like coded 
with Relative Extrusion - the rest remain in Absolute Extrusion mode.

Kind Regards,

Mark Kahler | Agfa
Support Engineer (PPS) | GS/Regional Services
T  +44 7771 974994 | M  +44 7771974994

Agfa Gevaert Ltd, Vantage West, Great West Road, Brentford, Middlesex TW8 
9AX, UK
https://www.agfa.com
Click on link to read important disclaimer: 
https://www.agfa.com/corporate/maildisclaimer/ 

(#7979)

nallath, 2020-07-01:
Well, this would indeed be something to do with a (custom) post-processing script. I don't think that enabling the relative extrusion per layer is a very good way to fix this. It would be really hard to make an understandable GUI for it and it's frankly a bit of a fringe case (that might even be printer specific)

closed 2020-07-01 by nallath

Ghostkeeper, 2020-07-02:
Probably best to just switch to relative in your customised snippet, and switch back to absolute at the end of your snippet then?