Ultimaker/Cura #7980: [4.6.1] UI shows wrong layer count on One at Time

LastDragon-ru, 2020-06-25:
**Application version**
4.6.0, 4.6.1

**Platform**
Win 10 x64

**Reproduction steps**
1. Add multiple models
2. Select Print Sequence: One at a Time
3. Slice 
4. Preview

**Screenshot(s)**

[image]

**Actual results**

Layers count = 330 + preview is not rendered properly for layers upper 240.

**Expected results**

Inside g-code I see

```
;LAYER_COUNT:165
;LAYER_COUNT:75
```

so layers count should be 240

konskarm, 2020-06-25:
I can reproduce the issue and will bring it up for discussion with the team.

Ghostkeeper, 2020-07-02:
I'm not able to reproduce this problem. I've loaded in a 1cm cube and a 2cm tall cylinder, selected one-at-a-time mode and with a layer height of 0.1mm I get about 300 layers in total, which is correct ((10+20mm) / 0.1mm = 300).

LastDragon-ru, 2020-07-03:
> Do you have a project file that reproduces the problem?

@Ghostkeeper, original models and settings

[Models.zip](https://github.com/Ultimaker/Cura/files/4868135/Models.zip)

[image]

Ghostkeeper, 2020-07-06:
Still not able to reproduce this problem.

You have one model which is 15mm tall. With your layer height of 0.15mm, this model will give 100 layers. The other model is 33mm tall, which gives 220 layers. There may be one or two layers off because the initial layer height is 0.1mm rather than 0.15mm, but this happens to even out in the resolution of the slice in this case. It gives exactly 320 layers:

[image]

However you didn't give me a project file or information on which printer you're using, which project you're using. Could you please click on File -> Save... and upload the resulting 3MF file here (in a zip)? This produces a modified 3MF file which also contains all of the settings you have adjusted, the printer and the profile you're using, and the precise placement of the objects in the scene. It should allow us to precisely reproduce the print.

LastDragon-ru, 2020-07-06:
> Could you please click on File -> Save... and upload the resulting 3MF file here (in a zip)?
Sure

[CFFFP_Handle.zip](https://github.com/Ultimaker/Cura/files/4879538/CFFFP_Handle.zip)

Ghostkeeper, 2020-07-08:
Ah, now it reproduces, using your project file. It seems to be dependent on the relative position of the models. If the taller object is printed first, this bug occurs. CuraEngine is slicing the correct number of layers but the front-end is displaying it wrong.