Ultimaker/Cura #7982: temprature in profiles

Alssund, 2020-06-25:
if I change a temperature in a profile, then shows it right in print options but not update the value in profile manager when I have pressed the update bottom

mahtDFR, 2020-06-26:
Please save a project file (File -> Save...), zip up the .3mf file and attach to this issue so I can try to reproduce it with the same settings as you.

Alssund, 2020-06-26:
I have made a document hope you understand it! (se attachement 
[temp differense.docx](https://github.com/Ultimaker/Cura/files/4838521/temp.differense.docx)

Alssund, 2020-06-28:
Now I have made a Custom printer profile, and the truble is gone!
It Look like it is when I used you pre installed "Anycubic Chiron" printer settings

nallath, 2020-06-29:
Check the value in the Extruder 1 tab. It should be correctly set there.

closed 2020-06-29 by nallath

Ghostkeeper, 2020-07-02:
The Extruder 1 tab will be empty, sadly, due to a different bug.