Ultimaker/Cura #7984: Prime tower materials not connected to build plate

BagelOrb, 2020-06-26:
**Application version**
4.6.1

**Printer**
Any dual extrusion printer

**Reproduction steps**
1. Load dual extrusion print
2. Enable prime tower
3. set brim extruder to left one

**Actual results**
layer 1:
[image]

layer 0:
[image]

layer 0 + 1:
[image]

total:
[image]

left extruder is on the outside on the bottom layer and on the inside on the rest of the prime tower.

**Expected results**
The materials are switched and connect through to the build plate, i.e.:
- The prime tower puts the [brim_extruder] material on the outside and the other on the inside
- The solid inside part of the prime tower doesn't cover the whole prime tower, but only the circle of the inner material
- Underneath the outer material there is also the outer material on the first layer

[image]

**Project file**
[some_dual.3mf.rename.zip](https://github.com/Ultimaker/Cura/files/4836118/some_dual.3mf.rename.zip)

**Additional information**
Related:
https://github.com/Ultimaker/Cura/issues/7906
https://github.com/Ultimaker/Cura/issues/7864

It is unclear whether my prints have failed because of this exact issue or another issue. My empirical evidence is suboptimal.

BagelOrb, 2020-06-26:
Maybe this photo proves one thing or another:
[image]

Ghostkeeper, 2020-07-02:
This is (currently) expected behaviour, not a bug.

The prime tower has an outer shell, which always gets printed on every layer. For this shell Cura chooses the material with the greatest layer adhesion, so that the tower is the most sturdy. All other materials get purged in the middle. They don't get printed every layer. With two extruders they get printed in at most half of the layers, with sometimes a big break in between. There is no way in which we can guarantee that it gets stuck to the bottom layer because it's not always primed in the second layer.

Instead we print the inside bottom of the prime tower with the same material as the shell. That way the shell is more sturdy. We purge the other material with a horizontal circle along the edge rather than a prime blob because it tends to stay a bit more neat, but it can't stick to anything because quite often there will be nothing beneath it to stick to.

The prime tower brim, by the way, is always printed with the Adhesion Extruder so that is unrelated. Without the prime tower brim your desired behaviour would only leave 3 line widths of adhesion for the sturdy outer shell of the prime tower, which is insufficient.

BagelOrb, 2020-07-02:
Indeed this is a feature request to solve a printing issue. I don't understand why you explain the current behavior in so much detail. It sounds like you are dismissing my feature request, but describing current behavior is not really addressing my feature request.

> Without the prime tower brim your desired behaviour would only leave 3 line widths of adhesion for the sturdy outer shell of the prime tower, which is insufficient.

3 line widths of adhesion is still more adhesion than zero line widths. If it's not enough then you shouldn't disable brim.

Just to be overtly clear; this is what I am requesting:
- with which material the shell of the prime tower is printed and with which the inside should depend on what the brim extruder number
- the flat base should always be the same material as the inside of the prime tower, not the outside
- the size of the flat base should be inside the normal prime tower of a layer, instead of covering also the outer material.

Ghostkeeper, 2020-07-02:
Ah, well with the "current behaviour" and "expected behaviour" I was interpreting this as a bug report. And I was indeed dismissing that as a bug because our expected behaviour is currently different, but kept the issue open for discussion.

Ghostkeeper, 2020-07-02:
We can consider the first improvement. I think it would indeed be an improvement for the prime tower. Needs some testing though.

With the second feature request I disagree for the reasons I mentioned before. There can be numerous insides of a prime tower (if using >2 extruders) and none of them are always printing on the second layer. Usually it will have many missing layers on the inside, making the extra adhesion from this feature only useful for 1 or 2 layers. It's more effective to adhere better to the outside shell which needs to be sturdy for the entire height of the build volume.
And since the flat base is the same underneath the shell of the prime tower, the third feature request is not applicable if we decline the second.

Is there a third opinion from a different developer, maybe, if BagelOrb are agreeing to disagree?

BagelOrb, 2020-07-02:
> Ah, well with the "current behaviour" and "expected behaviour" I was interpreting this as a bug report. And I was indeed dismissing that as a bug because our expected behaviour is currently different, but kept the issue open for discussion.

Yeah I guess I went with the wrong template here. Still this template is quite descriptive, so I like it.

I don't think we disagree, we just have different insights into it.

>  It's more effective to adhere better to the outside shell which needs to be sturdy for the entire height of the build volume.

I thought the two materials of the prime tower were symmetric. If the first material is missing for some layers then the sturdiness should come from the second material and the other way around. However, often you use the second material for support and then all layers have both materials (up until some layer).

> There is no way in which we can guarantee that it gets stuck to the bottom layer because it's not always primed in the second layer.

Indeed we cannot guarantee. We can only do our best, but it would be better with the proposed feature. Actually we could guarantee if we would also implement the option to always print the prime tower even if it's not needed on a given layer, but that's not what this feature request is about.

Ghostkeeper, 2020-07-06:
The two materials are not symmetric, no. It chooses one material as the outer shell (the one with the greatest Adhesion Tendency setting, or the first extruder among them if there are ties) and prints that material on every layer. All other materials are only primed on layers where they start halfway through the layers. So with 1 build material and 1 support material, it'll print the build material (most likely) as the outer shell every layer, but it'll print the support material every other layer.

Indeed we can do our best with details like this, but in this case it is detrimental to the platform adhesion of the outer shell of the prime tower, which is a greater goal in my opinion.

BagelOrb, 2020-07-06:
Ok I didn't realize that. I guess my request is not so pressing.

The current approach optimizes the printability of the one material, but minimizes material use of the other materials at the cost of printability. Depending on the materials we might want a prime tower where the inside is also printed every layer and has a brim on the inside.

On another note: if you look at the photo above you'll see that the prime tower now has a brim which doesn't provide anything to the prime tower, because the materials don't stick together. Would be nice if we could enforce that the brim of the prime tower is always the same material as the outside of the prime tower.

I think we should move away from the paradigm of generating brim lines from offsets from the parts on the build plate, and move to defining areas which we want to have filled with concentric brim lines. That way we can easily generate a brim for each material separately and have them not overlap using boolean operators on those areas.

Ghostkeeper, 2020-07-08:
Having brims with lines that can overlap stuff (or more accurately, prevent overlapping even if they would overlap) is something we've been wanting for a while. A common bug report is "Brim Only On Outside" still generates a brim on the inside if there is an island inside a hole. That would fix that bug too. I'm open to the idea as long as we can discuss this with more developers first (anyone?).

Alternatively we could also change the extruder for the outer shell of the prime tower to match the adhesion extruder instead of the one with the greatest layer adhesion setting. That is assuming that the adhesion extruder also has good layer adhesion. Might not always be true, of course, because people sometimes use PVA as adhesion extruder to have it dissolve more easily.