Ultimaker/Cura #7986: [4.6.1] Change retraction distance, speed and extra prime amount by height or with modifier meshes

Hirsute, 2020-06-26:
**Is your feature request related to a problem? Please describe.**
When calibrating settings for a printer after a part change (hot end change, nozzle change, sometimes even filament color/brand etc), the process is much quicker when an object can be set up to vary relevant settings in different parts (eg - Temp towers using the Change at Z post-processing script). I would love to be able to change retraction settings on different parts of a print to help me tune retraction after such changes. 

I checked the Change at Z script, and retraction settings are not available for modification there, which would be handy.

I also checked the per-object settings using a "support blocker" object placed over part of my stringing test, then set the blocker object to "overlap" and "cutting mesh", but retraction settings other than "prefer retraction" are not available there either.

**Describe the solution you'd like**
My preferred option would be to see the retraction settings made available in the Change at Z script (Retraction distance, retraction speed, and extra prime amount), as this would let me clearly, quickly, and easily set the heights on a stringing test object at which to change the retraction, making it very easy to determine the best settings from the result of the print. It also avoids having to change back and forth among the different settings over the course of the print, because it would only change at height.

An alternative option would be to see the retraction settings made available in the per-object settings. I thought of, perhaps, putting multiple stringing test tower sets on the build surface at once, and using the per-object settings to change the retraction settings for each individual one, which opens up possibilities for more complicated testing variations, but takes more time and effort to set up (especially because grouping the support blocker meshes, for example to help make moving them around more simple, causes them to fall below the build plane and then get stuck as objects that drop to the build plane if raised back up). I'm also not sure how complicated in terms of gcode or how hard on a printer it might be to be changing back and forth among different retraction settings repeatedly.

I would happily use either solution, or any other solution to accomplish this in the slicer, however.

**Describe alternatives you've considered**
The only alternative I can think of right now is to keep doing what I've been doing: printing a single object at a time, changing the settings after printing. This works, but it requires me either to sit right next to the printer to remove objects from the print bed and start new prints, or I lose a lot of time in the iteration. It's quite inefficient.

**Affected users and/or printers**
Anyone who is recalibrating retraction settings after changing filament, hot ends, extruder styles, nozzle sizes, etc.

**Additional context**
I can't think of any particular additional context that would be helpful. 

Hopefully I haven't missed this being asked before, I did search, multiple times, without finding anything. My apologies if this is something you've addressed in the past here. Thank you for your consideration!

Ghostkeeper, 2020-07-02:
Changing retraction settings per-object is not currently an option for us. Retractions can also happen when travelling from support to support, from prime tower to support, from one object to a different object, from prime tower to an object, etc. So there is often no context as to which object we should get the setting from or it's ambiguous.

Adding it to the Change at Z script should be possible though. That is a duplicate of this issue: https://github.com/Ultimaker/Cura/issues/7296.
Note that making it easier to tune profiles for printers is typically not something we're spending a lot of time on. There is a much broader audience for features that make Cura easier to use for engineers that want to print something with existing profiles. We're open to make the change but it's not getting a whole lot of priority.

closed 2020-07-02 by Ghostkeeper

TheDeejster, 2023-01-04:
found this after a google search, would just like to add that it would be nice to be able to change them in the "modify settings" type mesh.