Ultimaker/Cura #7989: last layer: move by jumping the obstactes (hole)

devildant, 2020-06-27:
**Is your feature request related to a problem? Please describe.**
for the last layer the movements are the same as on the other layers, if the model has a hole, the nozzle will bypass it which creates areas to be completed later, this has the effect of leaving traces due to change of direction and resumption of joining.

note: with ironing on it could make a surface really clean

**Describe the solution you'd like**
there should be an option for the nozzle to skip the holes and continue filling instead of going around the obstacle, to avoid having to complete the areas at the end, the last layer would be much more beautiful

**Describe alternatives you've considered**

**Affected users and/or printers**
all

**Additional context**

Ghostkeeper, 2020-07-02:
I'm not quite sure what you're requesting here. Is it something that a Z hop could solve? Or is it about the order of the top surface skin lines?

devildant, 2020-07-02:
Hello,

no, the Z hop does not change the displacement when filling the bottom / top layer.

I try to schematize you the current behavior and the desired behavior.
1) current behavior: 
when filling the top layer, when the cura encounters a hole, it goes around it
[image]

2) desired behavior:
instead of going around the hole, the hotends should jump over the obstacle and continue the line
[image]

is it more understandable?

devildant, 2021-05-23:
Hello,

by searching a little, prusa implemented this function with name monotonic infill
link : https://github.com/Ultimaker/Cura/issues/8909

Ghostkeeper, 2021-06-04:
Ah, indeed. I must've missed your response at the time last year.

I'll close this as a duplicate of #8909 then.

closed 2021-06-04 by Ghostkeeper