Ultimaker/Cura #7991: Certain Settings are Hidden(IE Retraction Settings) by the active machine when not supposed to be

DGxInfinitY, 2020-06-28:
**Application version**
4.6.1(Currently Latest)

**Platform**
Linux(elementary OS 5.1.5 Hera)

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Add Ender 3 Pro as only(default) printer.
2. Enter Print Settings, See that Retraction Settings are not visible, Only Print Temp and Build Plate Temp is visible
3. Go into Preferences > Settings Visibility
4. Try to enable other Material Settings(Settings other than Material Settings also have this problem)
5. A message next to certain settings "This Setting has been hidden by the active machine and will not be visible"

**Screenshot(s)**
[image]
[image]
[image]

**Actual results**
Settings will not be visible after enabling just as the message explains.

**Expected results**
Settings that are applicable should be shown and functional.

**Project file**
[Stadia Controller Mount.zip](https://github.com/Ultimaker/Cura/files/4842954/Stadia.Controller.Mount.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4842958/cura.log)

**Additional information**
I have tried PrusaSlicer and all the respective settings are available with the same Printer Setup(IE this isn't a Printer limitation or rather it shouldn't be). I get extremely good results with any other slicer when I import my settings. This isn't a printer related problem. When printing a file using any Cura profiles (default or imported) I get the same stringing issue and more worrisome is the blobs I get on new layers. This is in my opinion9could be wrong) because no matter what profile you set, retraction settings are being either overridden or rather disabled entirely. This result is very poor print results. I have seen many others have great results using older Cura versions and similar profiles. I do believe this is a bug as this new version simply doesn't give me the options and controls I have seen previous cura versions having. For your information I do not have my printer setup over USB or Octoprint, I simply load all GCODE on the SDCard and print that way. I do plan on doing an octoprint setup so we will see if that makes any difference. I have also reported this issue on the Ultimaker forums here: [https://community.ultimaker.com/topic/29917-help-setting-not-visible/](url)
My comment is still waiting for moderator approval but this issue is extremely annoying but I am sure we can get this resolved quickly! Thanks for you ongoing efforts to provide us freeloaders with exceptional quality and features!

DGxInfinitY, 2020-06-28:
Update: Version 4.0 Has none of these same problems. All options are available and work just fine. I haven't tried any other versions so I don't know what version it works up to.

nallath, 2020-06-29:
I strongly doubt that the settings that you are showing are available in PrusaSlicer, since they are settings that are specifically introduced for the Ultimaker Material Station. As such, those settings are only read by the firmware of ultimaker machines (and not Cura). That is also why Cura shows them as disabled.

closed 2020-06-29 by nallath

DGxInfinitY, 2020-06-29:
Ok so after further searching the retraction settings are now moved to travel. So between v4.0 and v4.6.1 the layout has changed. So Indeed PrusaSlicer does have the settings I am referring to, it doesn't, however, have the retraction settings as referenced in the Material Section(as those parameters are machine-specific and are only supported in Cura as of now). So all in all I should have gone through all the available settings before opening an issue. Derp.

Ghostkeeper, 2020-07-02:
Those settings are only used by Ultimaker's Material Station, the automatic filament loader. There's a brief explanation for it in the [settings guide](https://github.com/Ghostkeeper/SettingsGuide/tree/master/resources/articles/material) of those mysterious settings.