Ultimaker/Cura #7992: Theme - Layer View : Default color settings result in clipped-out details

daT4v1s, 2020-06-29:
**Application version**
4.6.1 and previous versions

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Ubuntu Linux - appimage, intel graphics

**Printer**
(Which printer was selected in Cura?)
Ultimaker 2+, probably not relevant

**Reproduction steps**
1. load stl & slice part
2. show preview mode and select "line type" dropdown
[image]

**Screenshot(s)**
[image]

[image]
the dimmer image uses max value of 240 and thus more details visible

**Actual results**
by default layerview in the theme.json files use 255(/255) as max highlight value
but somehow, maybe due to rendering of lighting results in values above 255 and get cut off,
removing details

when override theme.json layer-view color scheme to a lower number than 255
screenshots still show values higher than specified
e.g. if set to 240/255 still see 242/255

**Expected results**
(What should happen after the above steps have been followed.)
not clip and thus show more details

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)
N/A

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
maybe not relevant but logs say `OpenGL Version:  4.5 (Core Profile) Mesa 20.0.8`

**Additional information**
(Extra information relevant to the issue.)
-

nallath, 2020-06-30:
I've set the colors to 90% saturation (so 255,0,0 is now 230,0,0)

closed 2020-06-30 by nallath

daT4v1s, 2020-07-16:
https://github.com/Ultimaker/Cura/blob/master/resources/themes/cura-light/theme.json
this an impovement, (approx as saturated as possible without clipping)
but it seems `layerview_support_interface` should be desaturated, (less often used option, easy to miss, tho)

https://github.com/Ultimaker/Cura/blob/master/plugins/SimulationView/SimulationViewMenuComponent.qml
the `GradientStop`s in `LinearGradient` maybe should also be desaturated,
and perhaps tweaking the gradient to be more perceptually uniform may also look better

commit 5055b530 (2020-06-30): Lower the saturation of colors in the layerview

Fixes #7992