Ultimaker/Cura #7997: Cura crosses perimeters although combing and avoid printed parts are enabled

PuddingBaer91, 2020-06-30:
Hi,

i checked both options and also tried to play with these settings but cura still crosses perimeters:

[image]

smartavionics, 2020-06-30:
Could you please provide the project file? Do File->Save and then zip/rename the .3mf file and attach to this issue. Thanks.

PuddingBaer91, 2020-06-30:
sure, here you go
[Perimetercrossing.zip](https://github.com/Ultimaker/Cura/files/4850120/Perimetercrossing.zip)

smartavionics, 2020-06-30:
It's a lot better if you use the not in skin combing mode...

[image]

PuddingBaer91, 2020-06-30:
yeah i know but why traveling out of the part? And it still crosses the perimeters. In S3D it stays in the part completly...

Ghostkeeper, 2020-07-02:
Looks like it's doing this because it's travelling from one piece of support to the other. It doesn't avoid the perimeters of the part because support is printed before the part is printed, so there is no part to avoid yet on that layer.

Setting combing to "Not in Skin" doesn't make a difference with these travel moves for me. I don't know why it would make any difference in Smartavionics' version unless he edited the behaviour of that. It shouldn't make a difference because there is no skin at all on these layers. "Within Infill" also doesn't fix this because the travel moves are from support to support. It doesn't go inside any part because there is no part on that layer yet -- only in layers below it.

I'll ask my colleagues if they have any idea on how we can avoid this, but it might be difficult to avoid parts from other layers if the ooze under a nozzle extends beyond one layer thickness.

Workaround for now is to enable Z hops and retractions between support. Takes a bit longer to print though.

PuddingBaer91, 2020-07-02:
Hi, thank you for investigation. I don't want a fast solution. I wanted to tell you this to get an even better slicer ;) I don't know how to get it programmed but what we need is: the travel from one feature to another (support to solid or solid to solid with gap) should always the place where the gap distance is the shortest to avoid ooze. Or a specified place like z seam position... Thank you guys ;)

mahtDFR, 2020-07-03:
After some discussion we consider this to be a fringe case, and as a result we've decided not to priotitize this. It's deferred, but open to external contributions.

Ghostkeeper, 2020-07-06:
> Hi, thank you for investigation. I don't want a fast solution. I wanted to tell you this to get an even better slicer ;) I don't know how to get it programmed but what we need is: the travel from [...]

It does that as long as combing is enabled, but in this case it's travelling from one piece to another without regard for hitting some object that was printed on the previous layer.