Ultimaker/Cura #7998: [4.6.1] Print Settings - text in input boxes invisible

Protazy21, 2020-06-30:
**4.6.1**
**Windows**

**Ender 3 Pro**

**Reproduction steps**
1. Installed Cura with additional plugins: Octoprint, Settings Guide, Sidebar GUI, Thingiverse Browser
2. Noticed the saved values are invisible in the input fields
3. uninstalled Sidebar GUI, Settings Guide - no change
4. reset to default settings by deleting cura folder, went trheough the initial set up - no change
5. uninstalled cura, deleted cura folder from roaming again, installed back, went through initial set up - no change

**Screenshot(s)**

[image]

**Actual results**
No visible text in any of input boxes

**Expected results**
predefined/saved values visible

**Project file**
N/A

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4850098/cura.log)
**Additional information**

Protazy21, 2020-07-01:
The text is visible today...

closed 2020-07-01 by Protazy21

Ghostkeeper, 2020-07-02:
Maybe a corrupt shader cache or something then.