Ultimaker/Cura #8003: 4.6.1 not displaying model

MrTeroo, 2020-06-30:
**Application version**
4.6.1

**Platform**
Windows 10 Pro v1903

**Printer**
Ender 3

**Reproduction steps**
1. Uninstall & reinstall Cura

**Screenshot(s)**
[image]

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

MrTeroo, 2020-06-30:
Cura no longer displays the model.
If I uninstall then reinstal it will display the first model I open correctly.
When I close then reopen Cura, it no longer displays the model.

nallath, 2020-07-01:
Please include the logs as per the bug template.

MrTeroo, 2020-07-01:
[cura.log](https://github.com/Ultimaker/Cura/files/4857742/cura.log)

Ghostkeeper, 2020-07-02:
It's crashing because it's unable to create the OpenGL context. What video card are you using? Does it support OpenGL, and which version?

Could you try to delete the Qt Shader Cache? It should be somewhere in `C:\Users\<username>\AppData\Local\`, probably the `qtshadercache` folder if that's there.

Ghostkeeper, 2020-07-02:
Sounds a bit like this issue: https://github.com/Ultimaker/Cura/issues/5296
There's a possible workaround listed at the bottom of that thread.

MrTeroo, 2020-07-02:
Thanks for the replies.
I emptied the qtshadercachefolder but it made no difference. I added the key to the config, it also made no difference.

My graphics card is Intel HD Graphics 460. Cura has been working fine for months before this.

I thought I would do another install. I force removed this time using Geek uninstaller.
Reinstalled and again it opens perfectly first time.

I took a copy of the log after this succesful open.

I then opened it again and the problem was back. I took another copy of the log.

Interestingly there is more in the first log than the second? 2nd log is 7162 lines long, all of which match the 1st log.

The 1st log has added lines and is 7370 lines long.

Any more ideas?
[1st opening of cura.log](https://github.com/Ultimaker/Cura/files/4866035/1st.opening.of.cura.log)
[2nd opening of cura.log](https://github.com/Ultimaker/Cura/files/4866038/2nd.opening.of.cura.log)

MrTeroo, 2020-07-02:
[image]
[image]

Ghostkeeper, 2020-07-06:
I guess that's caused by the log rotation? The "1st log" contains one additional start-up of Cura.

I don't really know how to debug this, then. I don't see the problem. The log file seems to think that there are multiple OpenGL contexts, that's our only hint. But I don't know what could cause this.

One last thing you could try is to go into the Preferences and enable "Force Layer View Compatibility Mode". That'll allow Cura to run with OpenGL 2 instead of the more modern 4.

MrTeroo, 2020-07-06:
Ok, thanks for your time.

I have uninstalled it now. I found PrusaSlicer which is much better.

Thanks again.

wvanderp, 2020-07-10:
Restarting the computer seems to fix the problem. even when restarting cura without restarting windows.

[cura.log](https://github.com/Ultimaker/Cura/files/4902633/cura.log)

Ghostkeeper, 2020-07-13:
Does the "Force Layer View Compatibility Mode" preference work for you, @wvanderp?

MrTeroo, 2020-07-13:
Thanks Wouter

I reinstalled and tried that and it did the trick.

A computer restart fixed the issue.

closed 2020-07-13 by nallath