Ultimaker/Cura #8004: Search for Profile Name / Filter Profile Names

FPrefect20, 2020-06-30:
**Is your feature request related to a problem? Please describe.**
With the capibility of changing many different settings users are creating lots of different print-profiles. Often many copies with small changes for different object sizes, use cases, filament types and sometimes even for different colors depending on the ingredients in the filament. 
Accessing one specific profile is very hard if the list becomes longer.

**Describe the solution you'd like**
With a “filter” / “search” textbox on top of the profiles list it would be easy to find our profile.
(as known from MS Outlook searching your email box for example)

**Affected users and/or printers**
Benefit especially for power users (users who often work with cura)

mahtDFR, 2020-07-06:
I've added this to the backlog: CURA-7571

fvrmr, 2021-06-03:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features from our community and from Ultimaker, we have decided to remove this feature request from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2023-07-13 by github-actions[bot]