Ultimaker/Cura #8005: Deactivate Combing Mode for the first Layer

FPrefect20, 2020-07-01:
**Is your feature request related to a problem? Please describe.**
The "Combing Mode" is a great feature. The advantages are obvious.
But the combing mode really only makes sense on top of already printed layers.
If you start with combing mode activated at base layer of the one part you get strings of filament traveling around with your hotend and probably destroying your object later on - of the other part the base layer gets ugly due to the chaotic additional filament tracks.
**Describe the solution you'd like**
Attribute [checkbox] “no combing mode at base layer”
-or-
Attribute [integer] “starting combing mode from layer number x”
**Describe alternatives you've considered**
possible, think about why these alternatives are not working out.)
**Affected users and/or printers**
Everybody and (esp. for optical reasons as mentioned above) all filament types are affected.

smartavionics, 2020-07-01:
My cura builds already feature this settings, if you wish to try them go to https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README.md file there before using.

FPrefect20, 2020-07-01:
Great! Thanks!
BTW: is it ok to leave bugs via this platform for Version "master" as well?
(there´s still the crash problem when a users profile is corrupted)

smartavionics, 2020-07-02:
If a problem exists just in my builds please report at https://github.com/smartavionics/Cura/issues but if it affects both my builds and the UM builds it would be better to report it here.

Ghostkeeper, 2020-07-02:
To prevent scarring (or "chaotic filament tracks"), disabling combing is not enough. Instead of combing it'll go in a straight line to the destination. It'd also need to Z hop then.

smartavionics, 2020-07-02:
Also in my release...

[image]

You can ask for z-hop on the initial layer(s).

mahtDFR, 2020-07-06:
I've added this to the backlog as CURA-7572

fvrmr, 2021-06-03:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features from our community and from Ultimaker, we have decided to remove this feature request from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

leoshusar, 2023-02-19:
I would also love this feature. When printing things made of many small complex parts, it does so much unnecessary travels over where once will be walls, it sometimes takes even more than 3 seconds of dancing into the position, instead of one fast 200 mm/s movement.
[image]

closed 2024-03-04 by github-actions[bot]