Ultimaker/Cura #8006: [4.6.1] Start gcode won't save indentation

CCS86, 2020-07-01:
Hi guys,

I am trying to use some conditional gcode in my Cura start gcode block. In order to make it work, I need to indent lines of code.

When I first add the indentation, using tab or a series of spaces, the output gcode is indented and works perfectly.

However, after exiting Cura and re-opening, the indentation is no longer there in my start gcode. It seems to have an issue capturing the indent with the format used to save the code.

Is there another way I can format it, to make sure it persists?

Thanks!

Ghostkeeper, 2020-07-02:
Hmm, no I don't think it's possible. This is indeed a bug. It arises from the format with which we save the start g-code in a profile. The config files use indentation to mean a line continuation. So your setting is stored something like this:

```
	machine_start_gcode = ;Some start g-code
	    ;Indented start g-code
	;More start g-code
```

Python's `configparser` strips the leading whitespace. I don't think there is a way to prevent it from doing that. To fix this we'd have to use escape characters for all of that whitespace or something.

smartavionics, 2020-07-02:
As a workaround, is it possible to have a character that gets replaced at the start of each gcode line, i.e. all leading ~ characters could be replaced with spaces in the output so:
```
normal stuff
~~~~indented stuff
more normal stuff
```

is output with the middle line indented with 4 spaces.

A hack, I know, but if there's no other way...

fieldOfView, 2020-07-07:
Could this be fixed elegantly now that Cura has its own alternative to configparser? @nallath 
https://github.com/Ultimaker/Uranium/pull/625

nallath, 2020-07-08:
I think it might actually be solved with the new config parser.

nallath, 2020-09-21:
It has been fixed by the config parser. Tabs are now retained with me.

closed 2020-09-21 by nallath

CCS86, 2020-09-21:
Thanks guys!