Ultimaker/Cura #8007: [4.6.1] No longer populating {material_print_temperature_layer_0}

CCS86, 2020-07-01:
Hi guys, 

I recently added some conditional gcode to my start gcode, and it is breaking cura's ability to populate these temp variables.

Here is my complete start gcode block:

`;//////// START GCODE \\\\\\\\

G90                                          ; abs pos
M82                                          ; abs ext
M106 S0                                      ; fan off

if {heat.heaters[1].current < 170}           ; Check for hotend temp
		M109 S170                            ; If cold, preheat
		G4 S20                               ; Dwell

M104 S{material_print_temperature_layer_0}   ; Set hotend
M190 R{material_bed_temperature_layer_0}     ; Set bed

G28                                          ; Home

M557 X20:190 Y10:180 S15                     ; Define mesh grid
G29 S0                                       ; Load bed mesh
M376 H3                                      ; Mesh taper

G1 Z60.0 F5000                               ; Lift
G92 E0                                       ; Zero extruder
G1 X20 Y20 F4000

M109 S{material_print_temperature_layer_0}   ; Wait for hotend

G1 X10 Y0 Z.3 F2000
G1 X50 E5 F1200
G1 X140 E11 F1200
G1 Y2 Z.1 F9000
G4 S2
G1 Y3 F9000
G1 X20 Z.07 F9000
G1 Z5 ;E

G92 E-.7                                      ; Comp for ooze

;///////////////\\\\\\\\\\\\\\`

But this line specifically seems to completely stop it from filling any of the temps in:

`if {heat.heaters[1].current < 170}           ; Check for hotend temp`

Is there a way to work around this?

Thanks!

nallath, 2020-07-01:
Yeah, I think that breaks pythons string format function since it's something that is between {} brackets. I'm not sure if there is an easy workaround for that.

Ghostkeeper, 2020-07-02:
Preprocessing maybe?
Not standard g-code this, though.

GregValiant, 2024-11-20:
Current versions (5.8.0 and up) allow logic in the Start and Ending gcodes.  I'll go ahead and close this.
From my StartUp Gcode in 5.9.0 (the syntax changed from 5.8.1):

{if 'TPU' in material_type}
M92 E119                   ; Steps/mm for TPU
{elif 'PETG' in material_type}
M92 E102                   ; Steps/mm for PETG
{else}
M92 E97}                   ; Steps/mm for PLA
{endif}
{if machine_center_is_zero}
M206 X-116 Y-121 Z0     ; Home offsets 'Origin at center'
{else}
M206 X-1 Y-6 Z0         ; Home offsets Origin LF corner
{endif}