Ultimaker/Cura #8008: [4.6.1] Build plate adhesion extruder setting does not switch to active extruder

reuelk, 2020-07-01:
**Application version**
4.6.1

**Platform**
MacOS High Sierra 10.13.6

**Printer**
Custom built printer

**Reproduction steps**
1. In a dual extruder setup, both extruders are enabled.
2. Set the build plate adhesion extruder to extruder 2. (Had skirt selected as adhesion type.)
3. Disable extruder 2 and only keep extruder 1 enabled.

**Actual results**
The build plate adhesion extruder does not switch to active extruder 1 when G-code is produced. Instead, it remains on disabled extruder 2, causing extruder 2 to heat up and print the skirt. The user will not know that this will happen (took me a while to figure out what was happening) since the extruder setting disappears when the second extruder is disabled.

**Expected results**
When disabling an extruder, if that extruder is set as the build plate adhesion extruder, Cura should switch the build plate adhesion extruder automatically to the active extruder.

Ghostkeeper, 2020-07-01:
I'm unable to reproduce this, using Cura 4.6.1 nor the current master (to be 4.7). These are my reproduction steps:

* Add Custom FFF Printer with 2 extruders.
* Give the second extruder some other material so that you can see the difference in layer view more easily.
* Set the adhesion extruder to 2.
* Load a model and verify that the adhesion is printed with extruder 2.
* Disable the second extruder.
* Verify that the adhesion is printed with extruder 1.

When I disable the second extruder, you can see the extruder setting briefly reverting to extruder 1 before it disappears (because those extruder drop-downs are not shown if you have only 1 extruder).

This is a video of what happens when I try it: http://dulek.net/work/extruder_reset_disabled.webm

reuelk, 2020-07-01:
I tried to reproduce it again with the steps I outlined above and realized I omitted a step when recalling what happened. If you change the material after disabling the second extruder, Cura alerts you that the adhesion extruder is different. I'm not sure why, since to my knowledge, materials do not include the adhesion extruder. If you tell it to discard the change and use the default, it will go back to the disabled extruder. It will do this with any material that I switch to. I've recording a video here:

https://drive.google.com/file/d/1TflNGKNecG1nxH3nFLkB_dYz3bWLrd4p/view?usp=sharing

And here is the resulting g-code file:

https://drive.google.com/file/d/1sIK3-ta8CSmwufNvldC2_Cb3sUQOn_pM/view?usp=sharing

Ghostkeeper, 2020-07-03:
> Cura alerts you that the adhesion extruder is different. I'm not sure why, since to my knowledge, materials do not include the adhesion extruder.

It asks you for any customised setting, not just for settings that would change by the change of profiles.

Looking at your video, I think the bug is happening when you have a custom profile that overrides the adhesion extruder (probably any extruder-type setting). Switching to that profile, whether the setting was customised and discarded or not customised at all, makes it use the disabled extruder. I can reproduce that bug now. Thanks!

mahtDFR, 2020-07-06:
This is on the backlog as CURA-7574