Ultimaker/Cura #8011: MONITOR GUI is a mess

PigeonFX, 2020-07-01:
Is this just happening for me or is the GUI for the Monitor tab screwed up for everyone? 

[image]

fieldOfView, 2020-07-01:
You seem to be using a plugin that might be the cause of the mess; none of the "Open fan", "Close fan", "Unlock motor", "Cool down", "SD File" and "Opening Print Job" buttons are provided by Cura itself, and the "Resume" and "Abort print" buttons also look different from the standard ones. Even the button next to the IP address is not a standard button.

The MKS Cura Plugin may need an update.

nallath, 2020-07-02:
As @fieldOfView says, this is not implemented by us, but by a third-party plugin. Other plugins don't have this issue.

closed 2020-07-02 by nallath