Ultimaker/Cura #8012: Can not set Y min or X min Using Cura 4.4.1 (Mooz 2)

MakerReveiws, 2020-07-01:
**Application version**
4.4.1 

**Platform**
ubuntu 20.04.

**Printer**
mooz 2

**Reproduction steps**
1. Tried to reinstall Cura multiple time but same thing happened over and over again
2. I contacted Ultimaker but they told me to post an issue on github.

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

nallath, 2020-07-02:
I'm sorry, what min Y or min X? Also, have you tried to do this with Cura 4.6.1?

MakerReveiws, 2020-07-05:
I don’t know how to install 4.6.1. I downloaded straight from gnome
software. If you can tell me how to do it on raspberry pi running Ubuntu
20.04 then I will try that way.

Ghostkeeper, 2020-07-06:
The Ubuntu software repository has a build made by a maintainer in the community. It's a user-friendly way to distribute the software, but not maintained by the developers themselves. We don't know in what state it is. It could well be outdated.

You can always download the latest version here on Github or on the ultimaker.com website. 4.6.1 is here: https://github.com/Ultimaker/Cura/releases/download/4.6.1/Cura-4.6.1.AppImage

Did you mean the "X Min" and "Y Min" of the print head shape in the machine settings dialogue? Be aware that they must always be negative, since they are relative to the nozzle position. The nozzle cannot be more left than the left-hand side of the print head, unless you have quantum tunnelling filament or something. Was that what you meant?

MakerReveiws, 2020-07-07:
These are the settings that I have gotten from the Dobot mooz book. I don’t
know what to do. I was simply following instructions. I did not know. I
will try the git hub repo. Thank You.

Ghostkeeper, 2020-07-08:
Those instructions are maybe outdated. In the past those text fields worked differently.

closed 2020-07-22 by no-response[bot]