Ultimaker/Cura #8014: [4.6.1] Some UI problems with Archlinux KDE

Titome, 2020-07-01:
**Application version**
4.6.1 AppImage downloaded in the Cura website

**Platform**
ArchLinux, KDE, i7-8700k, GTX1080

**Printer**
Longer3D LK4 Pro

**Reproduction steps**
Open setting

**Screenshot(s)**
[image]
[image]

**Actual results**
Some texts are not visible or croped.

**Expected results**
See texts.

**Project file**
Don't need

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
I see similar issues but not exactly same #7300 #5490 #4826

I try to run from source but it's very painful :(

Ghostkeeper, 2020-07-06:
Duplicate of https://github.com/Ultimaker/Cura/issues/5490.

closed 2020-07-06 by Ghostkeeper

Titome, 2020-07-06:
I belive in #5490 the reporter use black theme in Cura but I'm wrong & MaxVerevkin's solution work for me for white label in white background : `env -u DESKTOP_SESSION -u KDE_FULL_SESSION -u XDG_CURRENT_DESKTOP ~/Downloads/Ultimaker_Cura-4.6.1.AppImage`

[image]

But the text still croped...

[image]

Ghostkeeper, 2020-07-07:
I also see the wonky text rendering, been there for a long time. It's still in the back of my head to look at one day. It seems to be specific to Linux.