Ultimaker/Cura #8015: Plugins not compatible?

ElePhoenix2020, 2020-07-01:
**Application version**
(The version of the application this issue occurs with.)
4.6.1

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?) Ultimaker S5

**Reproduction steps**
1. (Something you did.) Added plugins for Cura
2. (Something you did next.) Uninstalled and reinstalled as Cura was always removing itself each time I closed my laptop down.  Then, completely removed Cura from anywhere on my laptop and began the fresh and new installation of Cura.  About a dozen times.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
This is when Cura works:
[image]
If I click sync, This is when it erases itself once I close my laptop:
[image]

**Actual results**
(What happens after the above steps have been followed.)
Cura erases itself, even the icon on my desktop.

**Expected results**
(What should happen after the above steps have been followed.)
I should be able to open my laptop clamshell and use Cura...but its gone, with dust trails of me downloading it previously, but those links no longer work.  So I completely erase anything to do with Cura, begin again, and only when I sync, does it erase itself once I shutdown my laptop

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)
I did add plugins.  Can you tell me if one is corrupt or another issue maybe?  Please help, been spending hours on trying to resolve on my own...thank you!

ElePhoenix2020, 2020-07-01:
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/4860296/DxDiag.txt)

This is before I added plugins to Cura....Once I add plugins, it erases itself when I restart my laptop.  Thank you!

fieldOfView, 2020-07-02:
The dxdiag.txt file does not tell us what happened. Please see https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload (which will be named cura.log).

ElePhoenix2020, 2020-07-02:
I can't find cura.log  So, Ive spent so much time on this and if I just don't add the plugins I downloaded, Cura works fine.  Are plugins on my hard drive?  Cus I dunno which plugin is corrupt.  How do I delete plugins?  
[image]

ElePhoenix2020, 2020-07-02:
All the issues began when I downloaded the plugins off Ultimaker Marketplace.  I added all of them from: https://marketplace.ultimaker.com/app/cura/plugins

fieldOfView, 2020-07-02:
> I can't find cura.log

Go to help->show configuration folder. Cura.log should be in the folder that opens.

ElePhoenix2020, 2020-07-02:
This is what I get:

[image]

I unlinked all the plugins I downloaded to my Cura from Ultimaker Marketplace.  I am still getting a message to sync.  I have installed MasterControl, FreeCAD, Blender and MeshMixer.  Could these be corrupted?  Should I try to sync?  Cus every other time I synced, Cura deletes itself.....then I have to reinstall Cura.

fieldOfView, 2020-07-02:
The file I'm after is the "cura" one below cura.cfg. Windows has chosen to hide the .log extension. I really can't tell what is going on without that file.

fieldOfView, 2020-07-02:
Best thing to do is to keep the window open, start Cura and then sync. If Cura goes away, it will have recorded what has happened to make it go away in that cura(.log) file.

ElePhoenix2020, 2020-07-02:
What do I do now? 
[image]

ElePhoenix2020, 2020-07-02:
OK, I will sync now and wait for your reply on what app I need to open the file

ElePhoenix2020, 2020-07-02:
[image]

ElePhoenix2020, 2020-07-02:
[image]

fieldOfView, 2020-07-02:
You seem to have the wrong file. I need the "cura" one, not the "cura.cfg". You don't need to open it, you need to drag it into a comment here on github so it will get attched and uploaded.

fieldOfView, 2020-07-02:
Regarding your last image, you should Quit cura, as suggested. Then start it back up again.

ElePhoenix2020, 2020-07-02:
I dragged Cura.cfg and the message from this site says it is not supported.

ElePhoenix2020, 2020-07-02:
I stopped Cura, and restarted

fieldOfView, 2020-07-02:
> I stopped Cura, and restarted

So it is still there then?

> I dragged Cura.cfg and the message from this site says it is not supported.

I told you above, that's the wrong file! I'm not interested in "cura.cfg". I need "cura.log", which your Windows shows as just "cura".

This will be my last message for today; it is 10:40 PM here; way past work time.

Ghostkeeper, 2020-07-06:
You'd need administrator rights to remove an application from your computer. Cura doesn't have administration rights, so it seems weird that Cura would be able to remove itself from your computer.

We'll need to see that `cura.log` (on your computer it shows as just `cura`) to see if anything unusual has happened. You should be able to drag that file here.

closed 2020-07-20 by no-response[bot]