Ultimaker/Cura #8016: Cura 4.6.1 Crashes and reboots as if it was a new install

VE3EZB, 2020-07-01:
Hi folks - Twice since I've installed Cura 4.6.1, (5 May 2020) it has crashed unexpectedly and when it relaunches, it has lost all configurations settings.  It restarts as if it was a new installation - all preferences have returned to the "fresh install" settings ie: no printer, customize start and end g-code gone, GUI environment defaults back to "Ultimaker Theme" (I had it set to Ultimaker Dark) etc.
I've been using Cura for about 2 years now without issue until this recent update. 

**Application version**
Cura 4.6.1 (installed 5 May 2020)

**Platform**
OS - Windows 10 Home version 1909 - Build 18363.900
PC - HP ENVY x360 Convertible - Intel Core i5-6200U @ 2.3GHz 2.40GHz
Installed RAM - 8.00 GB
64-bit os x64-based processor

**Printer**
Creality Ender 3

**Reproduction steps**
1. Not sure I can reproduce the issue, it happened unexpectedly (twice) but the most recent time this happened (today) it seemed to be right after I had sliced and saved a file, then I pressed ctrl-N for a new project. I got a blue screen report from the OS saying something similar to "Oops something went wrong ..."  the Laptop reboots and when I relaunch Cura it is as if it was a new installation as mentioned above.

I have mentioned this in a Creality Facebook group and at least a few others seem to have had the same or similar issue also.

**Screenshot(s)**
Nothing to show - I get a blue screen, laptop reboots, and I relaunch Cura. 

**Actual results**
Cura 4.6.1 launches but has "forgotten" my configuration settings

**Expected results**
Cura works the same as it has always done in the pass - well

**Project file**
NIL

**Log file**
Nil

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-07-06:
[A Blue Screen of Death](https://en.wikipedia.org/wiki/Blue_screen_of_death) is caused by an error in the Windows kernel. It's typically indicative of a hardware fault in your computer, like a broken RAM stick, although it can also be caused by bugs in your drivers if they run in kernel mode, like your graphics drivers.

I suggest that you update all of your drivers (at least the graphics drivers and ethernet drivers, but if you can also the motherboard firmware). If the problem remains, you're most likely looking at a hardware problem.
Cura does not run in kernel mode. It is unable to trigger a BSOD, unless it happens upon a bug in the graphics driver that causes a BSOD.

When the computer hits a BSOD, it's unable to continue writing Cura's configuration files. It's possible that the configuration files got corrupt then and Cura would no longer recognise them. It'll see it as a factory reset then.

I don't think there is anything that we can do to help you with that BSOD...

VE3EZB, 2020-07-06:
Thanks for the reply Ghostkeeper.  The issue seems to have been with Microsoft Win10 updates.  Finally, after a "day long" upgrade from 1909 to 2004 everything seems to be working well. 
Cheers,
Steve

VE3EZB, 2020-07-06:
Issue resolved.

closed 2020-07-06 by VE3EZB