Ultimaker/Cura #8017: Center Camera When Item is Selected is not working

netenabled, 2020-07-02:
When trying to use this feature it is not working for me on a Macbook Pro OS Catalina Cura 4.6.1

mahtDFR, 2020-07-03:
Before I take this further, can you confirm you've enabled this setting in your general preferences?
[image]

netenabled, 2020-07-03:
Thanks for the reply, yes I can confirm that

[image]<https://nam10.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F17821912%2F86455568-17421a80-bd21-11ea-8f96-d121a8812ed4.png&data=02%7C01%7C%7C88eaf20fc40646eb828408d81f34490c%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637293656701168708&sdata=%2FXxaP69G%2BoIPRtRMOxU9LrPu%2B%2BiVo%2FXQG%2BpvyLGeUu4%3D&reserved=0>

mahtDFR, 2020-07-03:
Can't seem to reproduce this, it works in perspective and orthographic view (but we were only able to try in Mojave).
Maybe someone else can try and report back here?

Ghostkeeper, 2020-07-06:
I think @evtrados was about to test this, right? Or was there trouble with obtaining that laptop?