Ultimaker/Cura #8018: Terrible user experience using when scrolling lists in Marketplace.

leoheck, 2020-07-02:
I am using `Cura 4.6.1` on Linux `Ubuntu 20.04`.
This issue happens since forever.

All the lists inside the Marketplace are so hard to scroll with 2 fingers with the trackpad.
These lists have some kind of acceleration. When I start scrolling these lists, it accelerates until the end. It is so hard to use, manage, and find useful plugins/materials/installed stuff...

Is it possible to remove this acceleration somehow?

mahtDFR, 2020-07-06:
Thanks for the report. There seems to be something going on with every qt scroll view, not just in this dialog.
There is also a trade off, if we try to make this less sensitive it makes it harder to use with a mouse scroll wheel, which is probably the more frequent use case. It's probably something to do with the speed and acceleration settings.

We discussed it this morning and can't give it priority right now, so it's deferred, but open to fixes from the community!

mejobloggs, 2020-07-23:
I notice particularly weird scrolling when I'm adding a new printer, and scroll down in that list.

I find it unusable with the mouse scroll-wheel, and have to click/drag the scrollbar down, but even then it acts weird.

Complete speculation but... It feels like its trying to scroll collapsed lists as if they were open...?

For example, scrolling down.... sometimes it speeds up and sometimes super slow, and perhaps that's got to do with how many items in the collapsed list?

Even just scrolling down to "Creality" to expand the Creality printers is very frustrating. Would be much worse if you where further down the alphabet.

For reference, I also have Cura 4.4x and scrolling works perfectly in that

Ghostkeeper, 2020-07-24:
You speculated that correctly, indeed. It's an issue we had also seen. It started happening when we fixed a bug that caused some printers to not appear in the list (which resulted in a blank line). It's hella annoying. Also off-topic for this issue though since this is about the Marketplace.

leoheck, 2021-04-25:
I am using Cura 4.9 now and the scroll on Marketplace menus is still useless. I don't understand how such a simple thing can behave so badly. I never saw anything like this. RIP UX.

Dids, 2022-05-17:
This is still an issue everywhere, in both Cura 4 and 5.

Does anyone happen to know if this is a consistent issue on all OS's, as I'm only using macOS, and it's definitely an issue there?

closed 2023-06-24 by github-actions[bot]

leoheck, 2023-06-25:
This still happens on Cura 5.4.0-beta.1

When you scroll the list with a single step, it scrolls down to the end of the list. This is horrible behavior. There is no way to use the Mouse scroll there. The user has to manually grab the scroll handle and move it upnwards/downwards.

Here is the requested screenshot. 
[image]