Ultimaker/Cura #8022: Nozzle size

ginmarpro, 2020-07-02:
When I use CR-10 X printer  mode I need use two extruder with diferent nozzle size bu I  can t change nozzle size for some extruder.  In ENDER 3 PRO mode in material settings I can change nozzle size.

mahtDFR, 2020-07-03:
Thanks for the report. This is probably because the nozzle size variants haven't been defined CR-10 X. We probably won't fix it as we're stretched with other priorities. The original contributor of this profile or someone else in the community could make a pull request and we can review it.

There's some documentation about it here:
https://github.com/Ultimaker/Cura/wiki/Profiles-&-Settings#variant

You can see the definitions here:
https://github.com/Ultimaker/Cura/blob/master/resources/definitions/creality_cr-x.def.json
https://github.com/Ultimaker/Cura/blob/master/resources/extruders/cr-x_extruder_0.def.json
https://github.com/Ultimaker/Cura/blob/master/resources/extruders/cr-x_extruder_1.def.json

Nozzle variants are here:
https://github.com/Ultimaker/Cura/tree/master/resources/variants

Ghostkeeper, 2020-07-06:
> won't fix it as we're stretched with other priorities

Actually we won't fix it because we don't have a CR-10 X printer to test with and don't have the means to create printer profiles for 200 different 3D printers out there on the market. Ultimaker only maintains the profiles for Ultimaker's printers. The rest is down to the community, so I'll close this. If there is ever someone with the time to create some nozzle profiles for this printer, feel free to send over a pull request!

closed 2020-07-06 by Ghostkeeper