Ultimaker/Cura #8024: Foreign language translation is not being done in the profile menu dialog.

joonghochoe, 2020-07-03:
**Application version**
Cura Master(or 4.*)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10 + Intel i5

**Printer**
(Which printer was selected in Cura?)
Any 3D Printer

**Reproduction steps**
1. Select Preferences > Profiles > Select any active quality
2. Displayed only English settings at right table

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
(What happens after the above steps have been followed.)
Even if I change the language, only English name is displayed.

**Expected results**
(What should happen after the above steps have been followed.)
If I change the language, I want to see the setting's name for that language.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)
don't need project file

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
don't need project file

**Additional information**
(Extra information relevant to the issue.)
In the DiscardOrKeep dialog, it is displayed normally in the translated language.
[image]

"self._I18n_catalog" processing in QualitySettingsModel.py doesn't seem to be perfect unlike in UserChangesModel.py.

mahtDFR, 2020-07-03:
Thanks for the report, will discuss

mahtDFR, 2020-07-06:
This is on our backlog as CURA-7575

Ghostkeeper, 2020-07-28:
This is fixed for Cura 4.7, through this commit: https://github.com/Ultimaker/Cura/commit/56fc5f9706c9d7e7d3f70a7fc166ad91aedbbc0e

closed 2020-07-28 by Ghostkeeper