Ultimaker/Cura #8025: 3D model not saved in project

BagelOrb, 2020-07-03:
**Application version**
master.
Have been having this issue for a month or so

**Platform**
Linux, likely any

**Printer**
Ultimaker S5

**Reproduction steps**
1. load model
2. save project as 3mf

**Actual results**
3D model not saved in 3mf (see logs)
3D model not loaded when loading the 3mf file
unzipping 3mf file reveals that there is no folder '3D' inside

**Expected results**
model saved in project

**Log file**
```
2020-07-03 10:45:28,998 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [151]: Writing to Local File /home/t.kuipers/Documents/PhD/RVS_printing_project/Prints/UMS5_eagle_warp_test__v12.2_fragmented_walls_wave_fill.3mf in binary mode
2020-07-03 10:45:29,042 - ERROR - [MainThread] 3MFWriter.ThreeMFWriter.write [184]: Exception: Error writing zip file
2020-07-03 10:45:29,042 - ERROR - [MainThread] 3MFWriter.ThreeMFWriter.write [184]: Traceback (most recent call last):
2020-07-03 10:45:29,043 - ERROR - [MainThread] 3MFWriter.ThreeMFWriter.write [184]:   File "/home/t.kuipers/Development/CuraApp/Cura/plugins/3MFWriter/ThreeMFWriter.py", line 169, in write
2020-07-03 10:45:29,043 - ERROR - [MainThread] 3MFWriter.ThreeMFWriter.write [184]:     savitar_node = self._convertUMNodeToSavitarNode(root_child, transformation_matrix)
2020-07-03 10:45:29,043 - ERROR - [MainThread] 3MFWriter.ThreeMFWriter.write [184]:   File "/home/t.kuipers/Development/CuraApp/Cura/plugins/3MFWriter/ThreeMFWriter.py", line 84, in _convertUMNodeToSavitarNode
2020-07-03 10:45:29,043 - ERROR - [MainThread] 3MFWriter.ThreeMFWriter.write [184]:     savitar_node.setName(um_node.getName())
2020-07-03 10:45:29,044 - ERROR - [MainThread] 3MFWriter.ThreeMFWriter.write [184]: AttributeError: 'SceneNode' object has no attribute 'setName'
```

**Additional information**
I am on my branch where I am testing several features, but I don't think that's related.
I've rebased all my branches on the latest master less than a week ago.

nallath, 2020-07-03:
You need to update LibSavitar.

closed 2020-07-03 by nallath

BagelOrb, 2020-07-03:
[image]

BagelOrb, 2020-07-03:
Maybe it's not linked properly. Lemme check.

BagelOrb, 2020-07-03:
Ok sorry for bothering you!

BagelOrb, 2020-07-03:
My script doesn't automatically recompile -_-