Ultimaker/Cura #8028: Top layers added even when set to zero for some bodies in print

emil-e, 2020-07-04:
**Application version**
4.6.1

**Platform**
Windows 10

**Printer**
BLV mgn cube

**Reproduction steps**
1. Set top layers to 0
2. Slice

**Screenshot(s)**
[image]

**Actual results**
Top layers are added for some bodies in the print.

**Expected results**
No top layers added.

**Project file**
[CFFFP_Diffusers^digitmodule.zip](https://github.com/Ultimaker/Cura/files/4872263/CFFFP_Diffusers.digitmodule.zip)

**Additional information**
Project file is attached.

smartavionics, 2020-07-05:
This doesn't happen for me using a development version of Cura so perhaps whatever was the problem has been fixed?

[image]

fieldOfView, 2020-07-05:
I can't reproduce it with Cura 4.6 either

Ghostkeeper, 2020-07-07:
Me neither. I noticed that you had a Linear Advance setting which caused a warning in my log, but also with that plug-in installed the result was the same. I tried several layer heights as you said but I can't reproduce the issue.

I notice though that the settings used while you made that screenshot are somewhat different than what is used in this project file. For one, you're using a skirt but the project file uses a brim.

Was the project file made while the bug was reproducing?

emil-e, 2020-07-13:
How about this one then? In this one, solid layers are added in the middle of the model rather than at the top but I'm suspecting it might be caused by the same thing.

[CFFFP_advanced diffuser.zip](https://github.com/Ultimaker/Cura/files/4913135/CFFFP_advanced.diffuser.zip)

emil-e, 2020-07-13:
[image]

smartavionics, 2020-07-13:
Hi, I just sliced this latest model using the same settings from your earlier project file and got no solid layers anywhere apart from the bottom 3 layers.

emil-e, 2020-07-13:
@smartavionics How about this latest project with the settings from that file then?

smartavionics, 2020-07-13:
The file you provided today contains only a model and no settings. You need to do File -> Save, not File -> Export.

emil-e, 2020-07-13:
I’m so sorry, I will get you a saved rather than exported file ASAP.

emil-e, 2020-07-13:
[CFFFP_advanced diffuser project.zip](https://github.com/Ultimaker/Cura/files/4914115/CFFFP_advanced.diffuser.project.zip)

Here you go!

emil-e, 2020-07-13:
Okay, trying it on my machine gives my a missing layer (and thus a solid layer added above that as a bottom layer) no matter what profiles I choose. I even tried a different printer (Ultimaker S5) but that didn't help.

smartavionics, 2020-07-13:
I just sliced that latest project and got no missing layers or solid layers above layer 3.

There must be some difference between Cura 4.6.1 and what I am using here.

smartavionics, 2020-07-13:
You could try with my Cura build. You can find them at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. My releases can be installed alongside the UM releases. Please read the README.md file there before installing.

Ghostkeeper, 2020-07-14:
I can see it happening in Cura 4.6 indeed with that last project:
[image]

It seems that this is caused by bugs in the Maximum Resolution algorithm. If I change the maximum resolution to 0.1, the problem disappears. This algorithm was reworked for Cura 4.7, so such bugs should be removed now.

Sorry for the hassle! Fiddle around with the Maximum Resolution a bit until you find a value that works. It'll be fixed in the upcoming release.

closed 2020-07-14 by Ghostkeeper

emil-e, 2020-07-14:
@Ghostkeeper Excellent, thanks for the feedback!