Ultimaker/Cura #8029: Cura 4.6.1. quits after startup screen

apoorv07baj, 2020-07-04:
**Application version**
Cura 4.6.1
**Platform**
Windows 10, MSI RTX 2060
**Printer**
CR10s Pro ( before the attempted reinstall)

**Reproduction steps**
1. Tried Reinstalling Cura, Display driver
2. tried switching to an older version of Cura

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
Nothing changes after any of the steps tried. the startup screens shows up and then goes. A cura icon stays in the lower right corner of the taskbar but just disappears as soon as you even hover your mouse over it.

**Expected results**
Cura Starts normally

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4872295/cura.log)

**Additional information**
logs are from after a fresh reinstall

nsarzyns, 2020-07-07:
This kinda sounds like an issue I had before. I have no idea if it's the same issue but here's what I did.

First let me ask, did you clear out everything from your system that is cura related? So everything you can find in AppData and such? I don't have an exact list of folder to delete but just checking.

Anyway, what I did was went into the AppData folder and then to Roaming\cura\4.6. (The 4.6 is what verison I'm using. Select the folder with the version you're using. There may be multiple folder versions there.) When you open up the folder, there should be a file called _cura.cfg_. Go ahead and open that with you favorite text editor. I hope it's Notepad++ or Atom ;) In that file, there should a section _[general]_ and in that section there may be one or two or more lines that begin with _window_. Delete whatever line begin with window under the general section. After that save the file and try reopening Cura. This fixed my issue.

What I think causes this, is closing Cura when the window is displayed on a non-primary monitor. This also goes for the market place as well.

apoorv07baj, 2020-07-07:
I couldn't figure it out so I did a system restore to when it was working thankfully that did the trick haha only major thing that system restore did was delete my antivirus. I will reinstall it today to see if that was the issue

mahtDFR, 2020-07-07:
If this happens, there could be something in your userdata that's not playing nice. A bad plugin, definition, config file, etc.
To check this you can [reset your user data](https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations).
Backup your data by zipping the relevant folder (e.g. 4.6) and then deleting the original. Open Cura again, and it will generate fresh user data. This can usually get you to a point where Cura will at least start.

Ghostkeeper, 2020-07-07:
Those warnings about the Marketplace happen for me too. They seem to be harmless, not sure what's causing them yet. I think this may be more relevant:

```
2020-07-04 13:54:12,245 - INFO - [MainThread] UM.Qt.QtApplication.startSplashWindowPhase [226]: The preferences file cannot be found or it is corrupted, so we will use default values
2020-07-04 13:54:12,248 - INFO - [MainThread] UM.Qt.QtApplication.startSplashWindowPhase [235]: The preferences file 'None' cannot be found, will use default values
```

Maybe the fallback we're using there (if the preferences file is corrupt) is not working correctly.

Ghostkeeper, 2020-07-07:
Deleting or renaming causes the same message to appear in my log, but doesn't cause Cura to crash.

AndiP1234, 2020-07-13:
Can you find the cura.lock file at C:\Users\USER\AppData\Roaming\cura\4.6?
Kill all cura.exe over the taskmanager and delete/rename the cura.lock.

Ghostkeeper, 2020-07-15:
With the lock file, things are going to behave a bit differently because I'm on Linux rather than Windows. On Linux, renaming or deleting a file will still allow an application to use that file if it had already opened it before the rename or delete. While on Windows the rename/delete is not allowed while an application has opened it.

I tried the following:

* Rename `cura.lock` to `bla.lock` while Cura was running.
* Delete `cura.lock` while Cura was running.
* During the start-up phase, kill all Cura processes, and then rename `cura.lock` to `bla.lock`.
* During the start-up phase, kill all Cura processes, and then delete `cura.lock`.

With all of these, the behaviour is the same: Next time Cura runs, it just recreates the lock file. No crashes.

If Cura is killed while the lock file exists, it's expected that Cura will wait for 10 seconds during the start-up to see if the lock file disappears. After those 10 seconds it will continue the start-up sequence.

TheLastGriffin8, 2020-11-20:
#8793 is my problem same as this i think. i found out if i uninstall nvidia game ready driver. it works fine bad thing is i kinda need the driver

MariMakes, 2023-10-19:
Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, and we've introduced a lot of changes since this report. I'll be closing this issue. I hope you understand.

You can download the latest version of Cura with a bunch of improvements and a new QT version here:
[https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)

Apologies for any inconvenience caused by our delayed response.
Thanks again, and please let us know if you run into any other issues 💪

closed 2023-10-19 by MariMakes