Ultimaker/Cura #8030: XY Shrinkage Compensation (as a persistent material setting)

Spire9, 2020-07-04:
**Is your feature request related to a problem? Please describe.**
I have a dimensional accuracy request...  All plastics shrink as they cool down from the the 235°C printing temp.  I am needing to remember to scale every model in the X and Y direction to compensate for this predictable material property.  For ABS this is about 100.5%, and for PLA/PETG it's around 100.3%.  It's important to note that a "Z Shrinkage Compensation" setting is not being requested as it's not a significant issue at this time.

**Describe the solution you'd like**
I would like a Material setting added called "XY Shrinkage Compensation" to automatically apply scale to my models in the XY direction.  

**Describe alternatives you've considered**
The current workaround is to **not forget** to apply scale to every model, every time I import it, and every time I re-import changes.  This is tedious, and forgetting to do it is a mistake that won't show up until I test-fit the parts.

**Affected users and/or printers**
Everyone will benefit, dimensional accuracy is awesome.

**Additional context**
Screenshot:
[image]

avidNewb, 2020-07-06:
I think horizontal expansion achieves what you are looking for.  I agree that it should be a setting in materials though and just made a feature request for just that.

Spire9, 2020-07-06:
No, the horizontal expansion setting is definitely not the solution.  First of all, it's based on MM, not percent.  And second, it SHRINKS holes smaller.  So it's very different than scaling the model in the X and Y direction by the material shrinkage percent.

avidNewb, 2020-07-06:
Horizontal expansion works for holes and also all polygons.  There are separate settings and the description is wrong (outdated).  It also has the ability to either shrink or enlarge holes and polygons depending on whether you use a negative or positive number.  I have not had your experience where percentage was the right choice because internal portions do not shrink in the same way that external portions do because they don't cool quickly.  I am able to get dimensionally accurate models using horizontal expansion regardless of material type

Spire9, 2020-07-06:
Nope.  Horizontal Expansion works by swelling the printable area.  So if you have a 200mm wide model, and apply a 1mm expansion, your all your holes shrink by 1mm too.  And any holes smaller than 1mm vanish entirely.  @avidNewb, if you are using that setting to compensate shrinkage, you are making a mistake.

avidNewb, 2020-07-06:
Well I have no issues using it.  Here's a tip:  If you set a 1mm expansion (which is a lot), also set a 1mm hole expansion (negative and positive are inverse between the 2 settings).  This will negate the expansion you use for polygons on holes.  Then calibrate holes with the expansion you need for them alone.  Say you need a 0.5mm larger hole, you would add 0.25 (because it works on radius and not diameter) to the 1mm you offset horizantal expansion from and set hole horizontal expansion to 1.25.

I am able to achieve very accurate models doing this within any tolerance that I need

Spire9, 2020-07-06:
I don't think I can talk to you anymore without insulting you.  Sorry.

avidNewb, 2020-07-06:
Sounds like a personal issue.  I hope my comments helped others.  Grow up.

fieldOfView, 2020-07-06:
For the record, with the Material Settings plugin from the Marketplace, you can set the initial layer horizontal expansion per material.

Spire9, 2020-07-06:
Thanks @fieldOfView for the info.  I added the plugin and checked it out.  Unfortunately model settings like Scale are not in the scope of the plugin.

mahtDFR, 2020-07-07:
Thanks for the request. A feature along these lines is actually already on our backlog (CURA-7118) and is planned for a future release.

Spire9, 2020-07-07:
I see 7118 is a closed issue about unprintable features.  It makes no mention of overhauling horizontal expansion into a percent based stretch.  If this has not been discussed yet, we can do that here.

Ghostkeeper, 2020-07-07:
What mahtDFR mentions is a reference to Jira, a work tracking system that Ultimaker uses internally. It is not related to the Github issue 7118.

Spire9, 2020-07-07:
Ah, gotcha.  So can someone clarify what functionality CURA-7118 will provide exactly?  @mahtDFR implies it's similar, but different.

mahtDFR, 2020-07-08:
@Spire9 It's too early to say too much definitively but it would be a choice offered to the user. There are still a lot of UX considerations to be defined.

Spire9, 2020-07-08:
Hopefully I can help.

As far as the controls, the two would appear separate.  The model's scale on the left of the screen would say 100%, while the Material setting "XY Shrinkage Compensation" would say 100.5%

But in the graphic area, the scale and Shrinkage Compensation would both scale the model. 
So a 230mm model with 100% scale setting and 100.5% Shrinkage Compensation setting would fill the build area by 231.15mm  Then at room temperature the part would shrink to the target 230mm.  Awesomeness.

Ghostkeeper, 2020-07-08:
We've just refined it a bit further this afternoon.

The basic idea of CURA-7118 (and subsequently this ticket) is to have a single scaling factor for X/Y/Z. While we know there are differences in how it shrinks in the Z direction, this is what was chosen as a start for now. In the interface it would show the model with its original dimensions, but CuraEngine would create g-code that is slightly larger. As a result, the disallowed borders of the build volume will also be moving more towards the middle to indicate that you'll need a bit more space to compensate for the shrinkage. For multi-extrusion we've defined a couple of rules as well but it all comes down to that we need to have a single scaling factor for the entire mesh group so that multiple materials still interlink correctly even if their materials have different scaling factors.

Ghostkeeper, 2020-07-08:
The Z not being separate is a concession we have to make in order to get it released. Ultimaker doesn't have infinite amounts of time to build it and to test things for all of their materials, so it'll have to do for now.

The packing will not be a problem, since it scales the entire scene together. Only when printing in one-at-a-time mode it will scale the individual groups of model individually. So if it's interlocking or packed closely together in the scene, it'll still be interlocking or packed closely in the print, and it won't overlap. And conversely, the distance between objects in one-at-a-time mode will remain the same so that the print head doesn't collide with previously printed objects. That's the idea anyway.

Spire9, 2020-07-08:
I agree, programmers don't worry about discovering the shrinkage values.
As for expanding all the models as one scene, that's not accurate because it acts as if there is material in the air spaces that shrinks the scene in the printer.  By scaling each model in the XY direction, you match reality perfectly.  But **please** don't force Z Shrinkage Compensation to be equal.

tlhintoq, 2020-07-21:
@Spire9 

I don't know if you've realized it yet but Material settings are basically pointless as they get overridden by profile settings.

It is not a case of profile gets applied, then the specifics for the material get laid on top.
I went round-and-round with a guy on the Cura dev team about that but basically, they aren't changing it.
He went so far as to say if the profile override were removed then the material value would take effect: But they present no UI for that, so you''d have to go dig in the blahblah XML to change it each time, every time you ran the job or some babble to that effect... In other words nothing anyone is ever really going to do in the real world.

So basically, the only thing Material settings are any good for is asigning a color for on screen display, and a cost per KG so you can estimate a job.  Everything else is lost to profile settings that override the material settings.

Spire9, 2020-07-21:
I probably worded that poorly, but my mockup photo is correct.  I meant as a profile setting in the material section.  I updated the title slightly to clarify that.  Thanks @tlhintoq

avidNewb, 2020-07-21:
> @Spire9
>
> I don't know if you've realized it yet but Material settings are basically pointless as they get overridden by profile settings.
>
> It is not a case of profile gets applied, then the [...]

This is not true. If you edit your profile and remove the settings you want to come from materials, then they are applied when changing materials.  I use material settings for extruder temp, bed temp, retraction distance,  and retraction speed without any modifications.  If you add the material settings plug-in then you can add many other settings and i do use that for horizontal expansion. 

I think the initial request for a material setting was better suited to avoid the need for many profiles.

Spire9, 2020-07-21:
Honestly, I don't care where they put the value in the UI.
I just hope they make it function exactly as described in the opening post.  Anything less isn't worth anybody's time.

Ghostkeeper, 2020-07-23:
I do expect that the Z scale would be implemented separately eventually. We just have to make incremental improvements.

Spire9, 2020-07-23:
Sounds good.  I will try and measure Z shrinkage on a large print, but honestly I expect it to be negligible.  X/Y shrinkage on the other hand is very significant when you print parts that need to fit over, in, and around existing objects.

iFrostizz, 2021-02-22:
Hi, I see that the feature is displayed but not actually usable. I would personally really need it and I think that the process is much better than adjusting the steps/mm in order to compensate the error. Because the dimensions would be off for any other filament that has another shrinkage ratio.

Ghostkeeper, 2021-03-02:
Yeah, there is some news. In 4.8 we implemented a shrinkage compensation feature. This feature was in the end hidden from the interface because of two issues.

It was implemented to scale X/Y/Z all at the same time because it would be easier for the process engineers to optimise the setting then. But fairly quickly the process engineers found out that the shrinkage compensation made the dimensions worse in some cases. They found that the shrinkage is not linear in the size of the model and also that the shrinkage is not uniform in the X/Y vs. Z directions (which we all already suspected).

The other issue is that the implementation had some UX issues. The scale factor was applied from the centre of the bounding box of the scene, so if you had objects spaced very far apart the model actually ended up in a slightly different spot as where you put it on the build plate. The collision shadow showed the actual eventual position of the print, so it would show a collision shadow slightly shifted. The alternative would be to scale each model individually from the centre of their own bounding box, but that would cause multiple models to no longer properly align if they are placed on the buildplate to intermesh.

The process engineers at Ultimaker are still trying to work out a better strategy to compensate than just linear and uniform scaling. I hope that we've drawn a lesson to not just always go for a minimum viable product, but people are often forgetful in that aspect... But at least the compensation is taken a bit more seriously from the dimensional accuracy side so I have faith that it'll be improved to a workable state soonish.
The UX problem still stands though. In some ways that is a more difficult problem.

iFrostizz, 2021-03-02:
I have personally used this feature recently after that someone told me how to activate it on the UM forum. Before this, I wanted to develop a plugin to only compensate the XY axis. What I wanted to do it that when you place an object on the build plate, the default scale being 100%, this setting would make it at 100 * scaling factor, and the same when using the scaling tool on the left. So my approach was to actually use the functions that are scaling the object.

Spire9, 2021-03-02:
I'm not finding any joy in this "I told you so" moment.

Ghostkeeper, 2021-03-09:
It's definitely a strong "I told you so" moment. Although we all knew it would eventually be necessary. There is also something to be said for developing in small increments though. In this case, too, since it allows us to see the UX issues more quickly.

Ghostkeeper, 2022-01-18:
This is implemented through https://github.com/Ultimaker/CuraEngine/pull/1543.

closed 2022-01-18 by Ghostkeeper

juzer99, 2022-03-12:
> **Is your feature request related to a problem? Please describe.** I have a dimensional accuracy request... All plastics shrink as they cool down from the the 235°C printing temp. I am needing to remember to scale every model in the X and Y direction to compensate for this predictable material property. For ABS this is about 100.5%, and for PLA/PETG it's around 100.3%. It's important to note that a "Z Shrinkage Compensation" value would be far smaller, and is definitely not equal to the XY one due to the nature of the 3D printing process.
> 
> **Describe the solution you'd like** I would like a Material setting added called "XY Shrinkage Compensation" to automatically apply scale to my models in the XY direction.
> 
> **Describe alternatives you've considered** The current workaround is to **not forget** to apply scale to every model, every time I import it, and every time I re-import changes. This is tedious, and forgetting to do it is a mistake that won't show up until I test-fit the parts.
> 
> **Affected users and/or printers** Everyone will benefit, dimensional accuracy is awesome.
> 
> **Additional context** Screenshot: [image]

Hello,
I inquired about the lack of this feature recently and received the following from nallath  (Team Ultimaker):
[Posted yesterday at 05:15 AM](https://community.ultimaker.com/topic/39911-material-shrinkage-automatic-compensation/?do=findComment&comment=304511) · Material shrinkage - automatic compensation ?
Cura already has this implemented, but due to some issues that we had with finding the right values for materials (in the sense that it worked for all models that we wanted to print with it), the setting is disabled in the frontend .

However, you can set it if you change the material files.
 
<cura:setting key="material_shrinkage_percentage">100</cura:setting> 

You can also set it for XY and Z separately
 
<cura:setting key="material_shrinkage_percentage_xy">100</cura:setting>
<cura:setting key="material_shrinkage_percentage_z">100</cura:setting> 

Could anyone point me to the location of those material files?
I checked C:\Users\UserName\AppData\Roaming\cura\4.13\materials - folder is empty

Ghostkeeper, 2022-04-02:
> Could anyone point me to the location of those material files?
I checked C:\Users\UserName\AppData\Roaming\cura\4.13\materials - folder is empty

Custom materials can be added to that directory, but that's only the custom ones. To change them for the built-in material profiles that ship with Cura, you'd have to find them in your installation directory, e.g. `C:\Program Files\Ultimaker Cura 4.13\resources\materials`.

pilot76103, 2022-04-03:
This setting is visible and easily set on a material by material basis using the Material Settings plugin.

psiberfunk, 2022-04-30:
@Ghostkeeper : it looks like the commit was only made to the 5.0 branch.. is it not in the 4.XX line ?

psiberfunk, 2022-04-30:
Note that this DOES show up in the 5.0 beta line: 
[image]