Ultimaker/Cura #8031: Filling Pattern is Ignored With Density Set to 100%

FPrefect20, 2020-07-04:
**Application version**
Cura 4.6.1

**Platform**
Windows 10 64bit

**Printer**
(not relevant - any printer)

**Reproduction steps**
1. Draw any sutible object (e.g. cube with 20 x 20 x 20mm)
2. Slice with cura using settings:
     - Top/Bottom Pattern: "concentric"
     - Infill pattern: "grid" (or Lines, or...)
     - Infill Desity: 100%

**Actual results**
The Settings for "Infill Pattern" is ignored; the pattern of "Top/Bottom..." is used.
If you change the "Pattern Density" to 99% the given pattern is used as expected.
(workaround)

**Expected results**
Use stated pattern settings.
(this might be useful for getting more stable objects)

smartavionics, 2020-07-04:
Some (most) infill patterns are not really amenable to printing at very high densities. Too many corners and crossovers, etc. If you really want 100% infill, then lines or zigzag are probably the only sensible options.

FPrefect20, 2020-07-05:
Sure - of cause all 3-dimensional filling patterns don't make sense when going to 100% infill.
The problem is that every filling pattern is ignored when selecting 100% infill.
(the pattern of "Top/Bottom pattern" is used for infill instead)

Ghostkeeper, 2020-07-07:
This is on purpose. If printing at 100% density, it's better to print the whole thing as skin, so it automatically sets the number of bottom layers to 999999. The reason, apart from what Smartavionics already mentioned, is that it prevents there being a border between the skin an the infill, which causes the print to take longer to print and introduces a weak point / fracture line, which is precisely what people normally try to prevent when printing at 100% density. That border is necessary to allow different line widths, different speeds and different patterns and such.
[image]

You can override this by setting the bottom layers to a more normal amount, like 6 or so. Or by using 101% infill instead (since it actually checks for being exactly 100%).

closed 2020-07-07 by Ghostkeeper

FPrefect20, 2020-07-08:
Thanks for the workaround! :-)