Ultimaker/Cura #8033: [4.6.1] Crash trying to preview extremely large slice

Block137, 2020-07-05:
**Application version**
4.6.1

**Platform**
Win10
CPU: AMD Ryzen 3 1300X
GPU: Nvidia GTX1060 6GB
RAM: 32 GB

**Printer**
Custom Tool Changer

**Reproduction steps**
1. Slice a model which results in very large G-code output. Small layer height, fine details, scaled up, etc.
2. Press Preview
3. Wait for layers to process

**Actual results**
Slicing finished successfully, G-code can be saved but when trying to preview, after "processing layers" finishes, Cura crash (disappeared from running tasks, without a trace)

**Expected results**
Show print path without crashing

**Log file**
...uploading

**Additional information**
G-code can be saved successfully if saved before going into preview.
Start happening when G-code size approaches 1GB.
CPU and RAM usage are well below 50%. Will update info for VRAM.

nallath, 2020-07-06:
How did it crash? You got the crash manager, it stopped doing anything or you got a "Cura.exe is not responding" message?

Block137, 2020-07-06:
Cura window just disappears. No error message. Nothing remained in task manager. As if I press X and closed it.

nallath, 2020-07-06:
Okay, that's odd. There isn't anything in the logs about it running out of memory (or well, any crash, really). The last thing that I can see is that it took 97 seconds to load the layerview and that's it.

Ghostkeeper, 2020-07-07:
I can imagine it still running out of VRAM if it allocates 5GB of VRAM at the same time. The Layers3D shader requires 62 floats of VRAM per line if I counted correctly. If we assume about 20 bytes per g-code line, that 1GB file contains about 50M lines of g-code. This would be 3.1GB of VRAM by itself. Not quite the 5GB that's necessary to make it run out, but more than the 20% measured.

If we can reproduce it, maybe we could run Cura through GDB or something to see where it's crashing. I've successfully previewed g-code files of 1.4GB before (takes a long time!), but my GPU is one generation newer (RTX 2070).

Block137, 2020-07-07:
FYI I got VRAM usage from HWMonitor it may not be accurate. What monitoring software do you recommend?

Ghostkeeper, 2020-07-08:
The problem with any monitoring software is that if Cura would request 12GB of VRAM and the GPU can't comply, then it never actually gets reserved on the device so you won't be able to monitor that.

Question is really what we can do about it. The only real solution is to reduce memory usage. The problem I'm pointing at is this bit, where it takes the simulation view data and just passes most of it on to the geometry shader:

https://github.com/Ultimaker/Cura/blob/141ad8ff1dcadeb98a530a858374ca7a63cb4c13/plugins/SimulationView/layers3d.shader#L18-L39

But I'm also just conjecturing here, based on what you're doing before it crashes. We don't know if VRAM is really the problem here or it's something else.

Block137, 2020-07-09:
If that's the case, I can't expect for a fix but additional logging entry will be useful.

Ghostkeeper, 2020-07-10:
Cura is unable to log anything itself since the application is closed down when it happens. But maybe Windows itself can be convinced to leave a log file: https://helgeklein.com/blog/2018/10/creating-an-application-crash-dump/?PageSpeed=noscript

closed 2020-07-30 by Block137

Ghostkeeper, 2020-08-03:
You can also try using compatibility mode for layer view by the way. I'd expect that to use less memory.