Ultimaker/Cura #8034: Cura Adding Top Layers to Infill where None Exist.

TJOBriens, 2020-07-06:
Cura is adding "Top Layers" INSIDE parts. I know this is not an issue with the model because I can make top layers set to 0 and the infill works as it should.  But this option also removes the top layers of the part that should be made.

Is there a way to stop "Top Layers" from being inserted randomly?

**Application version**
4.6.1

**Platform**
Windows 10 
4790 core i7
RTX 2060

**Printer**
Anycubic Mega X

**Screenshot(s)**

Nuisance internal Top Layer Paths added for no reason:

[image]

Removed Top Layer by setting to 0mm thickness:

[image]

**Actual results**
See Screenshot

**Expected results**
See Screenshot

**Project file**

[Planter Footer.zip](https://github.com/Ultimaker/Cura/files/4876177/Planter.Footer.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

smartavionics, 2020-07-06:
It's not a bug, it's working as it should but if you want to remove those skin areas, increase the value of the skin removal width setting to, say, 1.

TJOBriens, 2020-07-06:
> 
> 
> It's not a bug, it's working as it should but if you want to remove those skin areas, increase the value of the skin removal width setting to, say, 1.

What "work" is this supposed to do? 
Why would anyone want an unnecessary separate path for the printer to come back and fill in?

It is a little better but it is still placing "Top Layers" inside. 

Should I increase it more?
Will increasing it have any adverse affects?

Thanks
[image]

smartavionics, 2020-07-06:
> What "work" is this supposed to do?
Why would anyone want an unnecessary separate path for the printer to come back and fill in?

It's because the walls are sloped and it's trying to provide a foundation for the wall on the layer above. Think about it, if the walls are sloped enough, the wall(s) on the next layer could end up above infill which would not necessarily provide good support so Cura puts skin underneath.

> It is a little better but it is still placing "Top Layers" inside.

No, that's something else, I think. Try turning off the fill gaps between walls setting.

> Should I increase it more?
Will increasing it have any adverse affects?

If you make it too big then narrow regions of skin that you really did want can disapear.

smartavionics, 2020-07-06:
BTW, you can save some print time by using 5% gyroid infill with an infill line multiplier of 1 and connect infill lines enabled. It will still be plenty strong, I think.

smartavionics, 2020-07-06:
You can save even more time by not using the sharpest corner z-seam setting, use user-specified, front-left (or whatever) instead. Now down to 11h3m.

smartavionics, 2020-07-06:
Furthermore, you're using combing without setting a value for max comb distance with no retract and that will produce under-extrusion so I recommend setting that to something like 10.

smartavionics, 2020-07-06:
Either that or just turn combing off completely. For models like this, combing offers very little advantage.

Ghostkeeper, 2020-07-07:
Indeed it's adding skin there due to the gentle slope of the wall. Cura will generate skin underneath the entire wall, down to the innermost wall. So if there is a gentle slope, it'll generate skin for 0.6mm (Bottom Thickness) underneath the wall as long as that is wide enough to print. Setting the Skin Removal Width to 2 line widths or so normally works well to reduce printing time and still prevent gaps from appearing.

See here the advantages and disadvantages of this setting:
https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/infill/skin_preshrink.md

closed 2020-07-07 by Ghostkeeper

TJOBriens, 2020-07-10:
> 
> 
> You can save even more time by not using the sharpest corner z-seam setting, use user-specified, front-left (or whatever) instead. Now down to 11h3m.

> 
> 
> Furthermore, you're using combing without setting a value for max comb distance with no retract and that will produce under-extrusion so I recommend setting that to something like 10.

The Gyroid Infill made too many obstacles inside for what I want to do. I want as little Infill as possible because I am planning on filling the inside with Sand is there a Tree like infill similar to Resin printing support? I would like to be able to fill the entire cavity with Sand or something else to make it heavy and reduce internal air because it is going into an aquarium.

Does this Look right for those other two settings? 

[image]

[image]

smartavionics, 2020-07-11:
Looking good.

Ghostkeeper, 2020-07-14:
> I want as little Infill as possible because I am planning on filling the inside with Sand is there a Tree like infill similar to Resin printing support?

You can try the Infill Support setting. There is no tree support like this though (it still prints the remaining infill with the infill settings); you'd have to actually hollow out your model.