Ultimaker/Cura #8035: Add horizontal expansion and flow rate to Material settings

avidNewb, 2020-07-06:
There are probably other settings that should be included as well but here is my request:

Horizontal expansion and flow rate vary from one material to the next.  Brands, colors, type, and even individual roll can effect these settings.  Currently I keep a spreadsheet and change these settings with each material change.  It would be beneficial if these settings were added to the material settings to be automatically set when we change material the same way that temp and retraction work.

Spire9, 2020-07-06:
As discussed here: https://github.com/Ultimaker/Cura/issues/8030 
horizontal expansion can not be used to compensate material shrinkage.  It swells the printable area.  So it has an opposite effect on holes.  Plus the proposed "XY Shrinkage Compensation" in the other thread would be based on percent.  So it would not need to be re-calculated for every.

avidNewb, 2020-07-06:
Found what I was looking for in the Material Settings plugin.  Keep in mind that you may need to delete any relevant settings from your profile to get them to sync with your material settings. Closed.

closed 2020-07-06 by avidNewb