Ultimaker/Cura #8037: Creating Material with other Materialtype results in non usable default Profiles

Achimobil, 2020-07-06:
**Application version**
4.6.1, but in the previous it was also

**Platform**
Windows 10

**Printer**
Creality Ender-3 Pro

**Reproduction steps**
1. Create a new Material and use a Material Type which do not exists as Generic. e.g. "PLA+" or "igildur" which are real existing material types for 3d Printer (see screen material)
2. Select the new Material in Ultimaker Cura.

**Screenshot(s)**
[image]
[image]

**Actual results**
You see the Material name is shown orange and the Profil shows "Not supported" but I can't find how to define materialtypes which are not existing as Generic types. see screen ProfilesMissing

**Expected results**
I would expect to see the default Profiles and all other self defined profiles like when I use PLA as Material type I can see the default and my self defined profiles I defined when I print PETG.

**Log file**
In the log file I start cura and change from a PLA Material to a "PLA+" Material and close Cura again. I hope there is enaugh information inside to see what the problem is.
[cura.log](https://github.com/Ultimaker/Cura/files/4881100/cura.log)

If you can tell me what I am doing wrong I will try to correct it but when it is an error please fix it, thanks
Achim

rburema, 2020-07-07:
Hi! If you want the Material Type to work, it needs to be either an existing material (, or one that you made from scratch, not recommended). What do those 'Sunlu PLA+ ...' profiles have as their Material Type? You can make your own material by taking that as the Material Type as well, or start from either PLA or Though PLA and tweak it until it's right for printing PLA+.

nallath, 2020-07-07:
You can find a bit more info here: https://github.com/Ultimaker/Cura/wiki/Profiles-&-Settings

So what basically happens is that you change to a material_type that does not have any matching quality profile. That is also why it says that it's not supported, but it's not actually preventing you from doing anything (just saying that you won't have as many pre-filled settings).

If you leave the material type to PLA, it will select the quality profiles of PLA.

Ghostkeeper, 2020-07-08:
Like others explained, this is intentional behaviour. And in fact, the whole purpose of Material Type is to select the set of quality profiles that are available when the material is used. If you want to share in the "default" profiles of PLA, use PLA rather than PLA+.

Sorry for the confusion! It should be documented better, but it's pretty advanced stuff for most users.

closed 2020-07-08 by Ghostkeeper

Achimobil, 2020-07-09:
PLA+ is a different Material than PLA. It contains additional things but can generally printed with the PLA Quality settings.
But e.g. From Sunlu you can get PLA and PLA+, and this is not easy to see, when everthing is PLA.

So how can I copy the things from PLA to use PLA+? I have not seen how to make that. 

And "iglidur" is a different material type. is high resistance material. For the 3mm that is available in the market place, but I have a 1.75mm printer and when I can't define these things, how can I get that working?

Ghostkeeper, 2020-07-10:
Let's just forget that the field says "Material Type". It's a vague term. What it effectively does is select which set of profiles to use. If you want your custom material to use the profiles that other PLA-like materials use, set it to PLA. If you want your custom materials to use the profiles from ABS, set it to ABS. Cura doesn't care whether your "PLA+" blend is actually anything like other PLA blends or not. It can consist of mercury and plutonium for all it knows, but as long as the PLA profiles work to print it nicely you should use the PLA profiles.

> So how can I copy the things from PLA to use PLA+? I have not seen how to make that.

You set the "Material Type" field to PLA. Your material will then share the profiles with other materials that also have the "Material Type" field set to PLA.

> And "iglidur" is a different material type. is high resistance material. For the 3mm that is available in the market place, but I have a 1.75mm printer and when I can't define these things, how can I get that working?

I'm afraid this is not possible. The profiles that are made for Iglidur are designed for 2.85mm filaments and a limited subset of the Ultimaker S5 printer's possible configurations. If you have an extruder train designed for 1.75mm filaments it'll not print nicely at all, because the heating track needs to be much longer for 2.85mm filaments, retractions can be slower, flow rate changes are harder, and so on. Cura will filter the materials on their diameter to match the accepted diameter of the extruder train, rounded to 1mm (so 2.0mm filaments will still be accepted even if the extruder train was built for 1.75mm, but 2.85mm won't).