Ultimaker/Cura #8038: no me permite visualizar la temperatura de la placa de impresión.

magalicervio, 2020-07-06:


Ghostkeeper, 2020-07-08:
[image]

magalicervio, 2020-07-08:
[image: image.png]
Buenas tardes, a mi no me sale, ni en ingles, ni en español. No se cual
será el problema. Lo desinstalé y volví a instalar, anteriormente tenía la
versión 4.1.1 y se visualizaba perfectamente.

Saludos

magalicervio, 2020-07-08:
[image]

fieldOfView, 2020-07-09:
[image]

closed 2020-07-09 by nallath

magalicervio, 2020-07-09:
Muchísimas gracias!!!!!