Ultimaker/Cura #8040: Bitmap import: translucency and linear mode swapped

michalsc, 2020-07-07:
**Application version**
4.6.1

**Platform**
macOS

**Printer**
Creality Ender 5 (does not affect the bug)

**Reproduction steps**
1. Create image with few shades of gray arranged in linear order or use mine: [Imgur](https://imgur.com/wbQVUiH)
2. Import the image with "Linear" mode selected. You will notice the heights do scale logarithmically.
3. Import the image for the second time with "Translucency" mode. The heights will scale in linear way.
The behavior of both import modes should be exactly opposite.

**Screenshot(s)**
Import settings and resulting profile (side view) in Linear mode:
[Imgur](https://imgur.com/jFHe1Xa)

Import settings and resulting profile (side view) in Translucency mode:
[Imgur](https://imgur.com/5JIZD7E)

**Actual results**
Linear import produces heights scaled logarithmically, whereas Translucency import produces heights scaled in linear way.

**Expected results**
Translucent import shall produce heights scaled logarithmically, whereas Linear import shall produce heights scaled in linear way.

michalsc, 2020-07-07:
Location of the bug: plugins/ImageReader/ImageReaderUI.py, lines 173-175:

```
    @pyqtSlot(int)
    def onColorModelChanged(self, value):
        self.use_transparency_model = (value == 0)
```

should be:

```
    @pyqtSlot(int)
    def onColorModelChanged(self, value):
        self.use_transparency_model = (value == 1)
```

michalsc, 2020-07-07:
Fix proposed in PR #8041

rburema, 2020-07-08:
Thanks for noticing & then immediately fixing it!

closed 2020-07-08 by rburema