Ultimaker/Cura #8043: [4.5.0] [Manjaro/Arch] Stuck on WelcomeScreen

LimitlessGreen, 2020-07-08:
# Stuck on WelcomeScreen

**Application version**
4.5.0 

**Platform**
- OS:          `Manjaro Linux`
- Kernel:     `Linux 5.7.0-3-MANJARO `
- Desktop:  `KDE - Plasma 5.18.5`
- Graphics: `Intel(R) HD`
- Resolution:  `1600x900`

**Reproduction steps**
Just start Cura initially

**Screenshot(s)**
[image]

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4889611/cura.log)
<details>
<summary>This is the console output:</summary>

```

/usr/lib/python3.8/site-packages/UM/PluginRegistry.py:4: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
/usr/lib/python3.8/site-packages/UM/View/GL/ShaderProgram.py:327: SyntaxWarning: "is" with a literal. Did you mean "=="?
  elif type(value) is list and len(value) is 2:
/usr/lib/python3.8/site-packages/UM/View/GL/ShaderProgram.py:329: SyntaxWarning: "is" with a literal. Did you mean "=="?
  elif type(value) is list and len(value) is 3:
/usr/lib/python3.8/site-packages/UM/View/GL/ShaderProgram.py:331: SyntaxWarning: "is" with a literal. Did you mean "=="?
  elif type(value) is list and len(value) is 4:
/usr/lib/python3.8/site-packages/UM/View/GL/ShaderProgram.py:333: SyntaxWarning: "is" with a literal. Did you mean "=="?
  elif type(value) is list and type(value[0]) is list and len(value[0]) is 2:
/usr/lib/python3.8/site-packages/UM/Settings/InstanceContainer.py:511: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if common_path is "" or not common_path.startswith(install_prefix):
/usr/lib/python3.8/site-packages/UM/VersionUpgradeManager.py:90: DeprecationWarning: invalid escape sequence \.
  ".*\.lock",       # Don't upgrade the configuration file lock. It's not persistent.
/usr/lib/python3.8/site-packages/cura/UI/ObjectsModel.py:65: DeprecationWarning: invalid escape sequence \(
  self._naming_regex = re.compile("^(.+)\(([0-9]+)\)$")
[MainThread] UM.Application.initialize [140]: Initializing Ultimaker Cura
[MainThread] UM.Application.initialize [141]: App Version 4.5.0
[MainThread] UM.Application.initialize [142]: Api Version 7.1.0
[MainThread] UM.Application.initialize [143]: Build type None
[MainThread] UM.Resources.__initializeStoragePaths [417]: Initializing storage paths
[MainThread] UM.Resources.__initializeStoragePaths [427]: Config storage path is /home/user/.config/cura/4.5
[MainThread] UM.Resources.__initializeStoragePaths [435]: Data storage path is /home/user/.local/share/cura/4.5
[MainThread] UM.Resources.__initializeStoragePaths [447]: Cache storage path is /home/user/.cache/cura/4.5
[MainThread] UM.PackageManager.__init__ [53]: Found bundled packages JSON file: /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager.__init__ [53]: Found bundled packages JSON file: /usr/share/cura/resources/bundled_packages/cura.json
Attribute Qt::AA_UseDesktopOpenGL must be set before QCoreApplication is created.
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [112]: Trying OpenGL context 4.1...
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [175]: Trying OpenGL context 2.0...
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [182]: We got at least OpenGL context 2.0: 2.1 No profile
[MainThread] UM.Qt.QtApplication.initialize [168]: Detected most suitable OpenGL context version: 2.0 No profile
[MainThread] UM.Qt.QtApplication.initialize [175]: Initializing job queue ...
[MainThread] UM.Qt.QtApplication.initialize [179]: Initializing version upgrade manager ...
/usr/lib/python3.8/json/__init__.py:293: DeprecationWarning: 'encoding' is ignored and deprecated. It will be removed in Python 3.9
  return loads(fp.read(),
[MainThread] UM.PackageManager._loadManagementData [161]: Loaded bundled packages data from /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager._loadManagementData [161]: Loaded bundled packages data from /usr/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.PackageManager._loadManagementData [179]: Loaded user packages management file from /home/user/.local/share/cura/4.5/packages.json
[MainThread] UM.PackageManager._saveManagementData [248]: Package management file /home/user/.local/share/cura/4.5/packages.json was saved
[MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [114]: Loading plugin configuration file '/home/user/.config/cura/4.5/plugins.json'
2020-07-08 11:10:35,493 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [444]: Loaded plugin ConsoleLogger
2020-07-08 11:10:35,535 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [78]: Found CuraEngine at: /usr/bin/CuraEngine
2020-07-08 11:10:35,537 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [444]: Loaded plugin CuraEngineBackend
2020-07-08 11:10:35,546 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [444]: Loaded plugin LocalContainerProvider
/usr/lib/python3.8/site-packages/cura/UI/CuraSplashScreen.py:73: DeprecationWarning: an integer is required (got type float).  Implicit conversion to integers using __int__ is deprecated, and may be removed in a future version of Python.
[... 537 lines cut ...]
  File "/usr/lib/python3.8/threading.py", line 302 in wait
  File "/usr/lib/python3.8/threading.py", line 433 in acquire
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 112 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f6ad9ffb700 (most recent call first):
  File "/usr/lib/python3.8/threading.py", line 302 in wait
  File "/usr/lib/python3.8/threading.py", line 433 in acquire
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib/python3.8/site-packages/UM/JobQueue.py", line 112 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap

Current thread 0x00007f6affdeb740 (most recent call first):
<no Python frame>
[1]    70447 segmentation fault (core dumped)  cura --debug

```
</details>

**Additional information**
I installed Cura from the official repositories.
Logfile and console output differ. (More infos on console)

LimitlessGreen, 2020-07-08:
Tried the latest 4.6.1 AppImage. Looks better:
[image]

mahtDFR, 2020-07-08:
Unable to reproduce the original issue here, anyone else able to try? I think we can consider it resolved since you have it working in the latest app image.

closed 2020-07-08 by mahtDFR

Ghostkeeper, 2020-07-09:
There was a problem with using the newer Qt version. Using the official AppImage builds works.

Termuellinator, 2020-08-30:
i've got the same issue on current manjaro ( KDE 5.73.0 / Plasma 5.19.4 / AMD Radeon RX 5700 XT (NAVI10, DRM 3.38.0, 5.8.3-2-MANJARO, LLVM 10.0.1), QT 5.15.0-4) with 4.6.2 installed from the repo
AppImage 4.7 works as intended it seems

Ghostkeeper, 2020-09-01:
During 4.7 we've fixed some things that Qt was giving us warnings about but that weren't actually giving us problems in the interface. Maybe some of those were actually giving problems with newer Qt versions then.