Ultimaker/Cura #8044: Brim per material

BagelOrb, 2020-07-08:
**Is your feature request related to a problem? Please describe.**
If you want to print with two warpy materials you can currently only print a brim with one material, which means that the other material has a high chance of lifting off the build plate and ruining the whole print.

**Describe the solution you'd like**
Have a separate brim for each material.
The one material should have priority in locations which are close to both materials, perhaps based on adhesion tendency.

**Describe alternatives you've considered**
Currently there is only the support brim which goes inward and this often does not provide enough adhesion.

**Affected users and/or printers**
All dual extrusion printers

**Additional context**
Made this request based on discussion in https://github.com/Ultimaker/Cura/issues/7984
Related:
brim extruder should follow objects https://github.com/Ultimaker/Cura/issues/7219
support brim on outside https://github.com/Ultimaker/Cura/issues/3824
disable brim per object https://github.com/Ultimaker/Cura/issues/6882

BagelOrb, 2020-07-08:
Options:
[image]
a) cut polylines
b) polygons
c) zigzag

BagelOrb, 2020-07-08:
A will practically be printed very similar to C
B can cause same small areas to disappear
C is quite hard to implement

My preference geos to A

BagelOrb, 2020-07-08:
All of the related issues can be solved by the following algorithm:
- generate brim lines (from outsides of ground areas only)
- do a polyline difference with all disallowed areas

For the 'first' material the disallowed areas are all ground polygons
For latter materials the disallowed areas are also where brim was generated with other materials

Ghostkeeper, 2020-07-10:
I can see two advantages of this:
* Better adhesion when printing materials of different types, especially if they don't adhere together well. An actual dual-extrusion print where they don't adhere well is rare, of course (because those print break apart easily they need to interlock and such) but if we make this work properly for support then the use case is much more common.
* The ability to interrupt the brim is something we've been needing for other feature gaps as well, such as the Brim Only on Outside setting that still generates brims on the inside if there is some island inside that hole.

A disadvantage is though:
* You'll either need to interrupt the brim lines or at least print the brim in smaller patches. Especially for a dual extrusion print these patches could be very small (like the [tree frog](https://www.thingiverse.com/thing:329436)). This will interrupt the flow a lot which could be detrimental to the adhesion again.

Ghostkeeper, 2020-07-10:
What about this alternative?

[image]

You'll still need to choose precedence for some materials over others, but only where there's an odd number of lines between two objects. Don't know how hard that would be to implement though.

BagelOrb, 2020-07-10:
A more easy alternative to what you propose is the following, where red gets a slight advantage of 0.4mm length:
[image]
You have to do the same proposed boolean operations as in the original post, but you have to do it sequentially for each offset.

fieldOfView, 2020-07-10:
Apart from the pattern to use, how how this work in the frontend? Specifically, how would it interact with the "adhesion extruder" setting?

BagelOrb, 2020-07-10:
> Apart from the pattern to use, how how this work in the frontend? Specifically, how would it interact with the "adhesion extruder" setting?

I was thinking of having an adhesion extruder setting which has the option [Not overridden] which means both are used.
Otherwise the old behavior will be applied.

Perhaps also a setting about Adhesion priority extruder. To determine whether green wins, red wins or its 50-50 in case of [Not overridden] which would then give the picture above.

BagelOrb, 2020-07-10:
In order to address the linked issues:
1. There should be a setting Enable brim, which can be tweaked per object.
2. There should be a separate setting for an outward support brim (the current 'support brim' is going inward)

Ghostkeeper, 2020-07-16:
There was a(nother) major advantage brought up.
Currently if you're using a dual extrusion printer but only use the second extruder for your print, you need to disable the first extruder or change the adhesion extruder to be the same as the extruder of your model. This is a bit of a UX-issue because people don't know that they need to change it or even that there is such a thing as a brim and you need to change the extruder for it. The punishment is that your single-extruder brim inadvertently still uses multiple extruders. We can't automatically change the adhesion extruder in our architecture now. But this is an alternative solution.

BagelOrb, 2020-07-16:
> There should be a setting Enable brim, which can be tweaked per object.

I've thought a bit longer about this and I think it would be better to make the existing setting Brim Width settable per object.

BagelOrb, 2020-07-16:
pseudocode for dealing with priority and the two options for non-overlapping brim lines.
```
for inset_idx in [0, max_brim_line_count] {
  for each extruder in ordered_on_priority {
    while inset_idx < extruder.brim_line_count {
      brim_line = extruder.first_layer.offset(inset_idx * w - .5w)
      brim_line = brim_line.differencePolylines(keep_out_zone)
      brim.add(brim_line)
      if priority == equal
        break
      else
        inset_idx++
      endif
    }
    keep_out_zone = keep_out_zone.unionPolygons(extruder.first_layer.offset(inset_idx * w))
  }
}

```

BagelOrb, 2020-07-16:
It should be possible to make Brim Width settable per object. First generate the excess of brim lines for the object with more brim lines, then generate the common brim lines ass offsets from the other object unioned with the outside of the inner brim of the former.

Ghostkeeper, 2020-07-17:
If it's settable per extruder it should be easy to make it settable per object.

> I was thinking of having an adhesion extruder setting which has the option [Not overridden] which means both are used.
Otherwise the old behavior will be applied.

This might mean that we have to implement something similar for the raft. For skirt and None it's not a problem. Raft will be something similar to the brim I guess.

To get it to split the brim along the middle as per the [latest suggestion](https://github.com/Ultimaker/Cura/issues/8044#issuecomment-656575564) we might need to use a Voronoi operation. Would make it more complex.

We've decided to add this to our backlog. Developers, see CURA-7598.

We still need to work out what the front-end will display as the disallowed areas, too. Perhaps we could even make the brim be interrupted by the edge of the build plate instead of having it added as border around the build volume.

BagelOrb, 2020-07-17:
> To get it to split the brim along the middle as per the latest suggestion we might need to use a Voronoi operation. Would make it more complex.

Not if you use my suggested algorithm. (See also my last pic above) It's actually pretty easy. You just generate one brim for each material alternatingly and make sure they don't overlap.

> We still need to work out what the front-end will display as the disallowed areas, too. Perhaps we could even make the brim be interrupted by the edge of the build plate instead of having it added as border around the build volume.

Good point! Perhaps we should keep adjustable brim per object as a separate issue then. First do per-material brim in order to keep things separate.

BagelOrb, 2020-07-24:
I thing we should use adhesion_extruder only for "Raft Extruder" and introduce a new setting brim_extruder. That way peoples project files will still use the raft they had selected, but the brim will default to be matching.

BagelOrb, 2020-07-24:
By using the alternating brim line generation in a way where you don't do the brim lines of each material the same amount of times you can even give more brim to the material which requires a wider brim proportionally.

Here an example where we alternate between generating two green brim lines and one red brim line:
[image]

BagelOrb, 2021-02-18:
I am currently unable to perform a print because I am not able to print a brim per extruder and always either object is curling up from the build plate. I therefore decided to try to implement this feature.

BagelOrb, 2021-02-18:
I propose the following settings changes:
- Enable raft
- Raft extruder
- Enable brim
- Brim priority

Replacing the old settings:
- adhesion type
- adhesion extruder
- skirt line count

Skirt line count is superfluous; it's functionality is better represented with Skirt/Brim minimal length.

Instead of setting the adhesion type to none, the skirt/brim minimal length should be set to 0. A skirt will only be generated if the total length of the brim with that material (if any) is less than the required minimum length.

Perhaps we should also introduce a setting:
- Enable skirt

One function of the skirt is to gauge whether the bed is level, so you might want to have a skirt even though only some of your objects already have enough brim, while other objects have no brim at all.

BagelOrb, 2021-02-18:
Whether there is a raft should be a global setting determined by the materials, so it should have a `"resolve": "any(extruderValues('enable_raft'))"`

The brim is set per object / per material and goes around only those objects.
If either material wants a raft then all brim settings should be ignored, because raft trumps brim.

The skirt is set per material and goes around the whole first layer.If a material has a brim then its skirt settings should be ignored, because who needs a skirt when you already have a brim?

BagelOrb, 2021-02-19:
According to one material expert (Tom), brim > raft > skirt. If one material asks for a raft and the other for a brim, then it's better to apply brims for each material then a raft.

But a material should be able to specify whether it would like a brim in case its preference for a raft is overruled. The material profiles should get
```
enable_raft = True
enable_brim = True
```
And the resolve function of `enable_raft` should be `all(extruderValues('enable_raft'))`.

Then for any other setting which depends on whether there is a raft or a brim, the raft trumps the brim, e.g. `"enable": "not resolveOrValue('enable_raft') and brim_enable"`. That is: any brim setting is only used if there is a raft, but not vice versa.

Ghostkeeper, 2021-02-25:
> According to one material expert (Tom), brim > raft > skirt. If one material asks for a raft and the other for a brim, then it's better to apply brims for each material then a raft.

Currently Cura has the order of raft > brim > skirt > none. So if any material asks for a raft, it's going to get it.

BagelOrb, 2021-06-17:
I've worked on this issue some months ago, but it halted because the whole setting inheritance system seemed to need a rewrite.

I did implement the basic functionality, though. Let's try to get the brim-per-material-functionality in Cura without overhauling the frontend.

I thought perhaps we can just have one checkbox setting: Brim Per Material. The backend then needs to make sure that the brim of each material is within the limit of the brim width of the Adhesion Extruder, while using the Brim Line Width of each extruder respectively (because different nozzle size). That way the impact on the frontend is minimized.

BagelOrb, 2021-06-17:
I would like to stress that I am currently wholly unable to do my multi-material prints if I don't model in these brims myself. This feature is a must-have for multi-material printing.

BagelOrb, 2021-06-17:
I can't find my implementation of the backend in the branches on github, so I'm afraid it got lost when we reinstalled my laptop not too long ago.