Ultimaker/Cura #8046: Coasting on sharp corner due to reduced flow

ahalekelly, 2020-07-08:
When printing a part with a sharp corner, Cura appears to reduce the flow, which makes sense. However, this triggers coasting, which causes a visible seam on the part. Is it possible to make a check to keep coasting from triggering in this situation? Maybe only coast before retractions?

See this example in Cura 4.6.1, the real seam is in the top right, but there's a second "faux seam" created in the bottom left which goes away when coasting is disabled.
[image]

Ghostkeeper, 2020-07-10:
I'm not able to reproduce this issue. This is what I'm getting: 
[image]

Could you provide a project file that reproduces the problem?

The reduction in line width before a sharp corner is due to the Compensate Wall Overlaps setting. You can try disabling that.

ahalekelly, 2020-08-05:
Sorry for the late response. Compensate Wall Overlaps does fix the problem, here's a project file demonstrating the issue though. 

[Switch Plate Top v105.zip](https://github.com/Ultimaker/Cura/files/5026792/Switch.Plate.Top.v105.zip)

Ghostkeeper, 2020-08-06:
That is just a normal 3MF file that you created through File -> Export... or by saving the slice result as 3MF model. You can create a project file through File -> Save...
Your 3MF file doesn't contain information about what settings and profiles you were using. Only the scene data itself. So if I load it, I don't get the same issues:
[image]

closed 2020-08-20 by no-response[bot]