Ultimaker/Cura #8047: Cura 4.6.1 - continuously hang

bonoboxita, 2020-07-09:
Cura 4.6.1 just installed after removing previous Cura 4.5

I'm using Win10 pro, 16GB ram, GTX1080, SSD, triple monitor configuration using Nvidia surround option.

The software is hanging /freezing for few seconds for ANY actions i do. If you change any printing parameters or move an object. 

I saw previous open case suggesting the issue was due to an overlay from the graphic cards software but i'm not using anything similar.

nallath, 2020-07-09:
Did you also have the issue with 4.5 or is this a new thing for 4.6.1?

bonoboxita, 2020-07-09:
no, 4.5 no issue.

Ghostkeeper, 2020-07-10:
Strange issue! Does the task manager indicate that Cura is using a lot of CPU power while it's freezing, or is it just idling at 0-10%? Is something else taking processing power, such as firewall software?

bonoboxita, 2020-07-10:
yes...CPU start to spike at 30% and triggering also the fan to increase speed.
It is annoying that even to write the layer height is hanging... i mean, as soon as you click the layer height box Cura is start to hang and slow down

Ghostkeeper, 2020-07-13:
I'm unable to reproduce this issue so far. It's like butter, on my computer.

Could you maybe share a log file then? There may be something in there that can be of help.

bonoboxita, 2020-07-18:
here the logs file
[cura.log](https://github.com/Ultimaker/Cura/files/4941101/cura.log)
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/4941102/DxDiag.txt)

nallath, 2020-07-20:
Oh, I see it already. You're slicing 62 objects at the same time. We've recently made a number of improvements regarding that. I also see some other things in the log that have gotten significant improvements for 4.7 

Changing the layer height is probably the most expensive setting to change, since alot of other settings depend on it. 

I'd suggest trying again with 4.7 and see if that solves the issues.

closed 2020-07-20 by nallath