Ultimaker/Cura #8048: [4.6.1] Unable to import Cura profiles

prnsmwmw, 2020-07-09:
**Application version**
4.6.1

**Platform**
MacOS 10.14.6 Mojave. 

**Printer**
Ender 5

**Reproduction steps**
1. Downloaded Recreus Filaflex 95A official Cura profile: https://drive.google.com/file/d/1cuz_Y7JYJX3N0N_fbNCMDljBSeux3Clm/view?usp=sharing
2. Imported in Cura
3. Got error message
"Failed to import profile from Users_/xxx.../FILAFLEX 95A MED-FLEX PROFILE 0.4mm NOZZLE._curaprofile:
<message>[Errno 22] Invalid argument </message>"

or 

1. Exported active Cura profile
2. Imported in Cura
3. Got error message
"Failed to import profile from Users_/xxx.../xxx..._.curaprofile:
<message>[Errno 22] Invalid argument </message>"
**Screenshot(s)**
[image]

**Actual results**
Cannot import profile. 

**Expected results**
Profile being imported successfully. 

**Additional information**
I am very new to 3D printing, so this might have a super obvious fix, but I did not find any solution when searching.

nallath, 2020-07-09:
The profile that you shared seems to be corrupted.

prnsmwmw, 2020-07-09:
Okay! I'll contact the manufacturers and ask for a working one!

Now it also seems I can export a profile from Cura and successfully import it, although that wasn't possible before. Who knows why. I restarted Cura, so maybe some weird cache thing?

closed 2020-07-10 by nallath