Ultimaker/Cura #8049: THE S5 MATERIALS STATION REPEATEDLY FAILS

programify, 2020-07-09:
***DO NOT REMOVE THIS ISSUE WITHOUT SERIOUSLY CONSIDERING A FIX.***

**Application version**
4.4.1

**Platform**
WINDOWS 10

**Printer**
S5 PRO BUNDLE

**Reproduction steps**
1. 6 REELS LOADED
1. NOTICE YELLOW (TRAP #3) NOT IDENTIFIED AS SUCH.
1. TRY TO PRINT SOMETHING.
1. NOTHING HAPPENS.
1. TRY AGAIN
1. PRINTER COMPLAINS ABOUT THE MATERIAL DOES NOT MATCH THAT CONFIGURED ON CURA.
1. NOTHING HAPPENS.
1. TRY SEVERAL TIMES.
1. NOTHING HAPPENS.
1. TOOK OUT FIRST THREE REELS
1. RELOADED ONLY YELLOW INTO TRAP #1.
1. TURNED EVERYTHING OFF AND ON AGAIN.
1. RECOGNISED YELLOW TOUGH PLA.
1. CURA RESTARTED.
1. TRIED TO PRINT
1. NOTHING HAPPENS.
1. SAVED PRINT TO DATA STICK WHILE CURA WAS BLIND PRINTER TO PRINTER.
1. WALKED OVER TO PRINTER USED DATA STICK. PRINTER STARTED PRINTING.

**Screenshot(s)**
S5 CANNOT TAKE SCREEN SHOTS

**Actual results**
THIS TEMPLATE SUCKS

**Expected results**
THE MATERIALS STATION SHOULD KNOW WHAT MATERIALS ARE LOADED IF THEY HAVE VALID NFC STICKERS.
SOMETHING SHOULD PRINT WHEN I TELL IT TO.
CURA FAILS TO ACKNOWLEDGE REEL NUMBERS #1...#6 - IF IT DID IT WOULD ALLOW A WORKAROUND - BUT YOUR PROGRAMMERS AND SYSTEM DESIGNERS AREN'T UP TO MUCH I GUESS.

**Project file**
EH?
**Log file**
EH?

**Additional information**
EH?

fieldOfView, 2020-07-10:
This is the issue queue for Ultimaker Cura. The Material Station is a very different product. If your issue gets closed, that would be because it is not something that can be fixed in Cura.

Also please don't use all capitals. It looks like you are screaming, and that's obnoxious.

Sidenote: I am not an Ultimaker employee; I am a 3rd party contributor to the project.

mahtDFR, 2020-07-10:
@fieldOfView is right for all of the above. I will pass your feedback onto their team but I will close this ticket as it's not relevant to Cura.

closed 2020-07-10 by mahtDFR

programify, 2020-07-10:
I suspected you wouldn't get it and sweep this under the carpet.

It is ALL to do with Cura...

Cura failed to keep up with current events at the printer.
Cura failed to let me override the design flaw built into the Materials Station.
Cura failed to warn me of the log jam of prints queued up for a printer that would never print again unless someone cleared the queue.
Cura lacks the ability to number the spool positions in the Materials Station.
Cura can't represent the materials present in the printer and has no way for human to tell it about the true status of the printer.

Cura is missing vital functionality because of flaws in the hardware its trying to control. Can you see now?

programify, 2020-07-10:
BTW THE WAY - I WAS SCREAMING AS CURA'S LACK OF FUNCTIONALITY IS OBNOXIOUS TO ME.

mahtDFR, 2020-07-10:
>It is ALL to do with Cura...

No it isn't, it's because of the machine's firmware. Noon'e sweeping anything under any carpet, it's just the wrong channel to report such things on. I've passed this on to their team.