Ultimaker/Cura #8050: NUMBERED SPOOL POSITIONS IN MATERIALS STATION

programify, 2020-07-09:
CURA TO ALLOW NUMBERED SPOOL POSITIONS IN THE S5'S MATERIALS STATION TO OVERCOME POOR HARDWARE DESIGN RELYING ON NFC (19TH CENTURY RADIO) TO WORK FLAWLESSLY.

nallath, 2020-07-10:
Cura doesn't even know about the material station, it just knows that there are a number of configurations that the printer supports, which is based on the extruder. To Cura, it really doesn't matter in what spool position the material is, only with what extruder it's associated.

The material station supports manual selection of the spools, if for whatever reason the NFC detection doesn't work correctly. You can do this by pressing the spool position icon on the machine and select "select type" or "load". In the case of load, if no nfc was detected, you can select a material from the list. 

Last but not least, NFC was first described in a patent released in 1983, which makes it 20th-century technology.

closed 2020-07-10 by nallath

programify, 2020-07-16:
Last but not least, radio itself is generally accepted I'm sure as a 19th century invention. It's hard to see how NFC could work otherwise. :o)

nallath, 2020-07-16:
It's also based on electricity, which is even older. I don't really see how that's relevant here.