Ultimaker/Cura #8051: [Feature Request] show if gcode file is already saved.

The-Force, 2020-07-10:
Sometimes after slicing a file I click the "save to removable drive" button and as it takes some time I change focus or I go AFK. Later, when I come back, the popup for "file saved - eject?" it's gone and maybe I'm not 100% sure I saved the file, so to be sure, I would click it again, wasting more time.

It would be great to know somehow if you already saved the file with the actual config.

Some possible solutions would be to just change the text for the save button to something like "save again" or "save unchanged", or maybe change the button color, or add * to the app window title when there are unsaved changes.

Ghostkeeper, 2020-07-13:
The message disappears after 30 seconds. I rather like that such pop-ups disappear automatically. Saves me having to click away all of those messages. Maybe something else can indeed be devised to show that.

mahtDFR, 2020-07-14:
Thanks for the feature request. We discussed it in our stand up this morning and decided that since it's a fringe case, we wont be implementing it.

closed 2020-07-14 by mahtDFR

fieldOfView, 2020-07-14:
Perhaps I can pick this up with the "Titlebar Info" plugin (https://github.com/fieldOfView/Cura-VersionInTitlebarPlugin). @The-Force, could you open an issue there, so I don't forget?

The-Force, 2020-07-15:
> Thanks for the feature request. We discussed it in our stand up this morning and decided that since it's a fringe case, we wont be implementing it.

Thanks for taking it into consideration.

> Perhaps I can pick this up with the "Titlebar Info" plugin (https://github.com/fieldOfView/Cura-VersionInTitlebarPlugin). @The-Force, could you open an issue there, so I don't forget?

Done.