Ultimaker/Cura #8058: Every version has the same bug. Check the attachment

AshuTJ89, 2020-07-13:
[image]

**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

mahtDFR, 2020-07-13:
Thanks for the image. Can you fill out the template please?

fieldOfView, 2020-07-13:
It is also not the intended design that the extruder boxes are below eachother.

It looks like you are using Windows, and printing via USB, right? What screen resolution and what DPI / scale factor do you use?

closed 2020-07-27 by no-response[bot]