Ultimaker/Cura #8059: Cura 4.6(.1) is not starting

AndiP1234, 2020-07-13:
issue: Cura is not starting, only shows the startup screen (package manager init) and frooze. The problem starts with Cura 4.6 while troubleshooting is uninstalled 4.6 and install 4.6.1 without removing the configfiles.
analyze: there is a file named cura.lock at C:\Users\USER\AppData\Roaming\cura\4.6
solution: After deleting the file cura is starting.

rburema, 2020-07-15:
> solution: After deleting the file cura is starting.

Thanks for reporting this. However, in the future this is unlikely to happen, since 4.6.2 has an upgrade script.

Since it's solved for you, I'm closing this now. If that seems incorrect to you, please reply, so we can reopen the issue if necessary.

closed 2020-07-15 by rburema

Ghostkeeper, 2020-07-15:
If the lock file still exists when Cura is started, it's expected that it'll wait for 10 seconds for that file to disappear. If it doesn't disappear after 10 seconds, it should continue booting.

AndiP1234, 2020-07-15:
In my case the 10 second timeout doesn't work.  Win10 64bit build1909, Python 3.8.3

KimmoHop, 2020-09-14:
Not sure if same thing but related:

I had Cura 4.7.0 open on Win10, when lights went out.
After power came back, Cura would start and jam. No CPU usage, no disk usage, just sitting there for many minutes. I tried to kill it and restart many times but the same result.

I looked for logs, but nothing there. Then I noticed that lock file time was before power out - removed it and Cura continued starting :)

When Cura is waiting (and not getting) lock, should there be a UI message for what is (not) going on, and maybe that in log file.

If one can give me a pointer what files to look at, I could do that. If the area needs compiling to test, then I have to pass - the build instructions are not clear enough (having built PrusaSlicer and SuperSlicer but not being able to build Arcus library to get to Cura Engine) :/

Ghostkeeper, 2020-09-15:
It is putting a message in the log file when that happens.