Ultimaker/Cura #8063: Settings window wont expand

luke112345, 2020-07-14:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fieldOfView, 2020-07-14:
Please fill in the issue template, or we can't even begin to understand what you mean, let alone fix it for you.

luke112345, 2020-07-14:
Aldo,

It’s very very simple, when I try to drag the settings window down to show all the print settings, it doesn’t move so I can only see one setting at a time. How is that hard to understand?

Luke

rburema, 2020-07-15:
@luke112345 For all we know you want to expand it horizontally, or meant something completely different with the word 'expand' than is normal in the English language. (Yes, we do get bug reports like that.) I know it seems overly pedantic, but it saves us a lot of  time, which we can spend actually programming.

Also, it isn't always about understanding the issue. Especially when it comes to UI problems like this, we need to know:
 - the OS
 - if there's something in the logs that makes us go 'ah, so _that's_ what's wrong'
 - what part of the settings you tried to drag to make it expand (the grippy points bar I assume)
 - preferably what state the window was in when this happens ... or does this happen all the time?
 - the current configuration ... like do you have two screens & change screen config often, or are you on a laptop (yes, this sort of thing has influenced the behaviour in the past)

As you can see, a lot of those would've been handled if the template had been filled in.

Otherwise the answer will just have to be 'Well, it works on my machine, can't reproduce'.

closed 2020-07-29 by no-response[bot]